niessner/BundleFusion #1: Building error

Attreyu, 2017-08-09:
I think that the problem has something to do with mLib - where exactly should the mLibExternals.zip file be extracted ? Are there other folders which need to exist besides External, FriedLiver and img ?

Error	1	error C1189: #error :  "Mixing a dll CGAL library with a static runtime is a really bad idea..."	d:\Kits\BundleFusion-master\FriedLiver\Include\CGAL\auto_link\auto_link.h	374	1	FriedLiver

Error	2	error MSB3721: The command ""C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v7.0\bin\nvcc.exe" -gencode=arch=compute_35,code=\"sm_35,compute_35\" --use-local-env --cl-version 2013 -ccbin "C:\Program Files (x86)\Microsoft Visual Studio 12.0\VC\bin\x86_amd64" -rdc=true -IInclude\cutil\inc -I"D:\Kits\BundleFusion-master\FriedLiver" -ISource\DXUT\Optional -ISource\DXUT\Core -IInclude\Uplink -I"C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v7.0\include" -I"C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v7.0\include"     --keep-dir x64\Release -maxrregcount=62  --machine 64 --compile -cudart static    -use_fast_math -DWIN32 -DNDEBUG -D_WINDOWS -DD3DXFX_LARGEADDRESS_HANDLE -D_CRT_SECURE_NO_WARNINGS -DNOMINMAX -D_UNICODE -DUNICODE -Xcompiler "/EHsc /W3 /nologo /Ox /Zi  /MT  " -o x64\Release\CUDACache.cu.obj "d:\Kits\BundleFusion-master\FriedLiver\Source\CUDACache.cu"" exited with code 2.	C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V120\BuildCustomizations\CUDA 7.0.targets	593	9	FriedLiver

Ivy147, 2017-08-11:
I met the same error MSB3721,too

tansangxtt, 2017-08-11:
I also met the same error MSB3721. For details, please see build log

[build log.txt](https://github.com/niessner/BundleFusion/files/1217239/build.log.txt)

Attreyu, 2017-08-13:
Error MSB3721 was gone after downloading the latest sources - Angela uploaded a couple of days ago.

Now I have a number of warnings related to GetVersionExW and GetVersionExA: was declared deprecated in DXUT.cpp and DXUTmisc.cpp respectively.

Another warning: "argument: conversion from double to float, possible loss of data" in SIFTMatchFilter.cpp

And finally, an error: Error	6	error LNK1181: cannot open input file 'FreeImage.lib'	c:\Users\a-drcost\Desktop\BundleFusion-master\FriedLiver\LINK	FriedLiver

I have that .lib in the mLib externals folder.

I still think it would be best to know the folder structure the developers are using on their machines, in order to compile this successfuly.

Also, when using Structure Sensor, how and what should we be installing on the iPad ?

Attreyu, 2017-08-13:
Build succeeded after downloading the latest version of the sources from here, after they were updated a couple of days ago.

Those using Structure sensor, on iPad - what do we need to install on the iPad ?

closed 2017-08-13 by Attreyu

A123456J, 2017-09-15:
@tansangxtt Hi, my error log is the same as yours, have you solved the problem? Thank you!

761687049, 2018-01-11:
my error log is the same

quanqhow, 2018-03-23:
My error log is also the same. And my environment is Windows 10 1709, CUDA 7.0 and VS 2013, the same error. Can you help me solve this problem?

761687049, 2018-03-23:
Using CUDA 8.0 maybe solve this problem.

quanqhow, 2018-03-23:
When I use CUDA 8.0 to compile this project, it can build the final executable file named FriedLiver.exe, but I cannot execute it. It pops out the error shows that "The application was unable to start correctly (0xc000007b)", what can I do for this issue?

761687049, 2018-03-23:
In the .txt file, Some parameters need to be changed, eg: Types of equipment. you can try it.

quanqhow, 2018-03-23:
Could you tell me how to do in details?

761687049, 2018-03-23:
I just find the .txt file recently, I haven't run it correctly also, because the Kinect1 always missing "*SDK2*.dll"，but Kinect2 isn't in the using equipments, I will try again. When I run it correctly, we can share more information.

quanqhow, 2018-03-23:
@Attreyu I run the FriedLiver.exe successfully at the first, but it stuck just a moment. I use the Kinect V1 for Xbox 360 to test FriedLiver.exe. My platform is windows10, CUDA 8.0. How did you test the application successfully in the past? Could you tell me how to do?