niessner/BundleFusion #2: Structure sensor - how to use ?

Attreyu, 2017-08-13:
I'm using Structure sensor, on iPad Pro 9.3.

How do I link the tablet to the workstation where Bundle Fusion is running ?

Attreyu, 2017-08-15:
Managed to figure out - for Structure you need to open the Structure app and tap on Uplink in order to connect to the workstation, while Bundle Fusion is launched - same way you can connect to Skanect Pro 1.9.

But there is a problem - it is crashing when trying to connect to the tablet - laptop with Windows 10, latest update, with mobile Geforce 860M - it is detected, then "waiting", then crashes.

The drivers are the Geforce one, not CUDA, because CUDA 7.0 drivers are not playing nice with Windows 10.

Any suggestions ? What configuration are you using for testing and deployment ?

hggaoq, 2017-08-16:
I'm also using Structure sensor, on iPad Pro 9.3.3.
I have successfully compiled the project with CUDA 8.0 on Win10, but when I try to connect the server through the Structure app's Uplink button, the app changes to Depth+Color mode automatically,and the server changes nothing.
It seems that the app does not connect to the server correctly. As my Structure app's version is 1.4,  I downgraded the app to 1.3.2, which is its previous version on app store and try again. The result is that the app told me to upgrade its version while clicking the Uplink button.
Does anyone knows where the problem is? 
PS: My Structure Sensor's Firmware Version is 2.0, and the Uplink module of Structure app 1.4 works well in Skanect Pro 1.9.

willfu, 2017-08-26:
Same issue, it seems the uplink server is out of date while tracing the uplink server code.

willfu, 2017-08-26:
@hggaoq could you send the 1.3.2 version app to me? I guess it could work. Thanks.

Attreyu, 2017-08-29:
Anyone here managed to use a Structure sensor for scanning ?

Angela/Matthias - could you please test the code with the latest version of the app ?

Attreyu, 2017-09-03:
Using a Structure sensor is still impossible - with crashing on trying to connect to the uplink of the app on iPad.

lettucehead, 2017-10-02:
Hey there -- i'm with Occipital's support team. Definitely reach out to us! support@structure.io will do the trick. Include the device's Serial Number please

Attreyu, 2017-10-02:
I was just having a chat about other things with Anthony Monaco while I received your message :) so he'll give you my email and we can continue offline ? Thanks for reaching out !

Attreyu, 2017-10-02:
Model ST01
Serial 19746
UPC 8 115770 2034 2

Attreyu, 2017-10-28:
So basically the solution is useless, since it doesn't work with the only wireless sensor of the list.

nufanDK, 2017-11-01:
Could you elaborate on that comment? I'm about to test it, but has the Structure support team told you that it won't work, or am I missing something here?

Attreyu, 2017-12-08:
@nufanDK - I mean it doesn't work, as simple as :) the Structure team are aware of this, they said that they'll investigate, but afaik nothing has been done.

Have you had any luck ?

nufanDK, 2017-12-08:
No luck for me either. The structure team have responded that it simply isn't supported.

> "Unfortunately, BundleFusion will currently not work with the Structure Sensor via Uplink, and since it is not an Occipital Product, we can not be of much help in terms of support."

I've started to look into getting the structure sensor working tethered to the pc instead as a primesensor, so that I can test bundlefusion at least. Then I can start looking into the untethered solution if bundlefusion works ok.
However, I'm stuck with an error for memory not allocated for a variable. Can't remember it exactly as I'm not at my machine right now. Must look into it during the holidays :-)

Attreyu, 2017-12-08:
Well, since in the youtube original video about BundleFusion, they were using an iPad, I assume it must've worked originally, maybe with an older firmware version.

nufanDK, 2017-12-08:
True. From what I could gather of info, it seems, the uplink.h and uplink.cc that was used in the video is obsolete, and doesn't work with the current generation of uplink. However, I've been unable to get the source code for the current uplink feature.

Attreyu, 2017-12-08:
Yep. Sadly, the guys behind BundleFusion seldom reply and the sources haven't been updated in ages, except some change in the readme file.
If they could look in here, I'm pretty sure that this issue could be fixed in a matter of hours by them, with a simple code change, followed by the publishing in here.

rayhero, 2018-02-06:
I meet the problem too. I can connect it by Skanect but fail on BundleFusion. I not meet the crash but It stuck on waiting for connection... just like @hggaoq . Does anyone found the way to connect bundlefusion with structure sensor using uplink??

electech6, 2018-03-19:
What are the correct steps to use Structure App for BundleFusion? @nufanDK I only change the  "s_sensorIdx = 7" in "zParametersDefault.txt",  then run the VS project and the it shows "waiting for connection ...".  Afterwards, I  open the structure APP, but there is no "uplink" button shown.  My iPad and PC are indeed in the same WiFi. It's confused because it appears "uplink" button and works well when I open the Skanect.

euzer, 2019-05-21:
Hello everyone,
Uplink work when  i am using Skanect on windows. I have just connected my laptop and the structure sensor on the same wireless network.
But with BundleFusion,  it is not working well, also i can see the uplink button appear on iPad.
any ideas ??

euzer, 2019-08-28:
> Beacuse it use another udp server method , we must know and slove it now!

That is correct ! But do you have any clue to do this ?

JohnnistWangc, 2019-08-28:
Beacuse it use another udp server method , we must know and slove it now!

JohnnistWangc, 2019-10-25:
Only the reverse app-side software transport protocol and find several layers of instructions of  the handshake protocol can simulate the authentication of the server side bypassing the app-side.

chethanab16, 2020-12-15:
@rayhero @willfu @Attreyu @electech6 @lettucehead  does any one successful  in working structure sensor with Bundlefusion , if yes could you some one suggest me how to solved the above problem,