niessner/BundleFusion #3: How to get output data?

tansangxtt, 2017-08-15:
I cannot find the scene after scanning, please give me some instruction. Is it in 2 config files?

tansangxtt, 2017-08-16:
I found it in DepthSensing.cpp. Your instruction should mention these functional keys

closed 2017-08-16 by tansangxtt

kuonangzhe, 2017-10-18:
@tansangxtt 
Hi, I am trying to make this code work, too. Did you got the colored reconstruction result? I made it run but the window 'Fried Liver' showing is dark and the output results are also blank images(dark). Do you have any suggestions? Thank you so much!

adammpolak, 2018-01-25:
@kuonangzhe what is your CUDA/VS/DirectX version?

BruceWANGDi, 2020-10-12:
@kuonangzhe  Hi, I have the same issue with you, did you solve this problem? My environment is CUDA8.0, VS2013, Win10, but my GPU is RTX2080ti, I don't know whether my GPU can work on CUDA8.0.