niessner/BundleFusion #5: Hardware  requirement

zdczdcc, 2017-08-17:
My computer only has one graphic card,GTX 560 Ti with 1024 GPU memory.I want to run the program after compiled it successfully.But it reports `CUDA error,invalid device symbol...`.So i change the `minDevices=1` in `DualGPU.h`,then it report `_debugger_hook_dummy=0 error`.I googled about it found that it's a memory error.
So i want to know is it possible to run the program on only one graphic card with enough GPU memory,or it is necessary to run with two graphic cards.
Thanks for your help.

hggaoq, 2017-08-17:
I'have tested for GTX 960(2G memory) with i7 4790, and it reports that No enough GPU memory, while it works on GTX 970(4G memory) with i7 6600k.

zdczdcc, 2017-08-17:
@hggaoq ,
Did you use only one gpu with 4g memory and where did you modified for it in the source code?

hggaoq, 2017-08-17:
Nothing but the config in zParametersBundlingDefault.txt.
s_maxNumImages 1200 -> s_maxNumImages 600

zdczdcc, 2017-08-17:
@hggaoq ,
I would get one graphic card with 4G memory and have a test with your config.Thanks for your help!

zdczdcc, 2017-08-17:
@hggaoq ,
I've tested for  Quadro K4200(4G memory)  with i7 7860.It also reports the error `cutilCheckMsg() CUTIL CUDA error, CUDASolverBundling::CUDASolverBundling  : (13) invalid device symbol...`.

My software environment is `Windows 7`,`VS 2013`,`CUDA 8.0(compiled error with CUDA 7.0)`,`Kinect SDK v1.8(uncomment the KINECT macro and set s_sensorIdx=0)`,and the others mentioned in the README.md.

And i've captured the data from `apt0.sens`,it also reported the same error.

Would you give me some advice?
As you said,you've got  an error that meaning No enough GPU memory and i wonder to know does it same as the `CUDA error,(13) invalid device symbol...`.

hggaoq, 2017-08-17:
My GTX960' error is also a CUDA error, not enough memory,which will be printed to the console.
My successful test is using s_sensorIdx=8, and the test data is a .sens file downloaded from this project website.
Until now, I can only run this code successfully with s_sensorIdx=8. Unfortunately, I failed with s_sensorIdx=0,1 or 7.

zdczdcc, 2017-08-17:
@hggaoq ,
I would have some debug with s_sensorIdx=8 firstly.Thanks for your help!

closed 2017-09-26 by zdczdcc

nicole-antoine, 2018-06-20:
If I may write a new comment, I'm encountering the exact same issue with a GTX 770, i7-4770 on W10. I tried with kinect sensor et their data.sens but it seems it has no link with the error.
Would you have any idea how I could correct this ?

calyx1991, 2018-11-30:
Hi, I have got the error：runtime API error 10： invalid device ordinal.  I'm wondering if I only have one GPU， how can I run this project successfully？