niessner/BundleFusion #6: Kinect XBOX One doesnt work

tansangxtt, 2017-08-17:
I have tested it on Kinect One but it doesn't work properly. DirectX window is back and not responsing. Found out the code, it stuck at this line:

	while (hr != S_OK) hr = m_pMultiSourceFrameReader->AcquireLatestFrame(&pMultiSourceFrame);
Line 41, KinectOneSensor.cpp

tansangxtt, 2017-08-17:
m_pMultiSourceFrameReader->AcquireLatestFrame(&pMultiSourceFrame) always returns E_PENDING

willfu, 2017-08-19:
It is not stucked at this line, you can add break point later than that loop.

tansangxtt, 2017-09-04:
@willfu Have you enabled KinectOne compilation flag?

dave101326, 2017-10-27:
I have the same problem, it seem that it only capture once.
 
Any idea?

[image]

yaoy908, 2017-11-09:
hello
I try with Kinect one, and the command window stucked after running. My graphic card is gtx 960m. Do you have any progress? I can run with sens file after ignoring "cuda out ot memory" error. But I cannot run it with Kinect one nor realsense r200.

yaoy908, 2017-11-09:
In my project, it stuck at line "ConditionManager::waitBundlerProcessedInput(ConditionManager::Recon);" in OnD3D11FrameRender.

Jaiy, 2017-12-01:
@dave101326 
Hi, have you solved the problem? I meet the similar problem as your picture shows, then my comment show the error——cudaSafeCall() Runtime API error 74: misaligned address. Could you give me some suggestions? Thanks a lot.

yaoy908, 2017-12-04:
@Jaiy 
Update your video card or decrease parameters in zParametersDefault.txt.For example, increase the size of voxelsize.

Jaiy, 2017-12-04:
@yaoy908 
Hi, you are right, when I increase the size of the voxelsize, the error above disappeared, thanks a lot first. But the command window still stucked after running, can you run the project successfully with the Kinect one? Have you solve the problem? Or can you run successfully with other device? Thank you for replying!

yaoy908, 2017-12-04:
@Jaiy 
I modify the sensfilereader class, to read my images from kinect, so I didn't try it with realtime-running.On my pc, kinectonesensor cannot work well. You can modify it, but my pc cannot affort a good fps with realtime-running, so I choose to read images from disk.

ljilekor, 2017-12-06:
Hello everyone

I'm using an old Kinect (s_sensorIdx = 0) on windows.

I got stuck at this error:
**d:...\bundlefusion-master\friedliver\source\depthsensing\MarchingCubesSDFUtil.h(58) : cudaSafeCall() Runtime API error 2: out of memory.**

The DirectX window doesn't pop up.

[image]

Does anyone have an idea on how to get around this error please?

Thanks in advance

Jaiy, 2017-12-07:
@yaoy908 
Hi, thanks for your kind help, I have solve the problem by changing the CUDA version, it's strange, but it did work. Thanks again!

carstenschwede, 2017-12-14:
@Jaiy Could be a bit more specific as to what you did to make it work with the Kinect One?

CDElite, 2018-01-18:
@Jaiy 
Which version of CUDA did you work with?

quanqhow, 2018-03-23:
I run the FriedLiver.exe successfully at the first, but it stuck just a moment. I use the Kinect V1 for Xbox 360 to test FriedLiver.exe. My platform is windows10, CUDA 8.0. How did you test the application successfully in the past? Could you tell me how to do?
[image]

quanqhow, 2018-03-24:
I also tested my Kinect V2 (for XBox One) under the FriedLiver.exe, but it still stuck at the moment. So I have tested both kinds of  Kinect and the FriedLiver stuck at the two conditions. Are there anybody able to help me to solve this tough issue?

junweiy2012, 2018-03-29:
@tansangxtt I met the same problem as yours. Also got stuck at 
while (hr != S_OK) hr = m_pMultiSourceFrameReader->AcquireLatestFrame(&pMultiSourceFrame);

Have you solved this problem?

quanqhow, 2018-03-30:
@junweiy2012  Sorry, I haven't solved this issue. Have you figured out some solutions?

junweiy2012, 2018-03-30:
@quanqhow try by commenting out the use_multi_thread option in GlobalAppState file. It works for me after commenting it out. I don't know why though.

callMePlus, 2018-04-24:
Hi everyone , I meet the same  problem : It always tell me "E_PENDING" or "E_FILE" when run at ```hr = pColorFrameReference->AcquireFrame(&pColorFrame);``` in the process of getting color frame.I found the constructor function solve such problem by using the ```while(hr != S_OK)```  . I followed it , and It's OK unexpectedly. This's what I did :
```
hr = pMultiSourceFrame->get_ColorFrameReference(&pColorFrameReference);
if (SUCCEEDED(hr))
{
	hr = E_FAIL;
	while (hr != S_OK){
		hr = pColorFrameReference->AcquireFrame(&pColorFrame);
		Sleep(100);
	}
}
```
I hope it can help you

nicole-antoine, 2018-06-19:
Hello, I'm currently trying to make this bundlefusion project work with KinectOne or Kinect v2 (since it seems not to work with structure sensor atm) but whichever I tried I got stuck with 
[image]

would you have any clue why ?

WuYingzheng, 2018-06-19:
@nicole-antoine 
hi,
    I got the same error. I have struggle several days to compile this project, and it succeed but fail to run. And i guess the reason is the CUDA compile parameter doesn't match you GPU. Change you CUDA compile parameter for a try. but i did't know if it works.

nicole-antoine, 2018-06-20:
@WuYingzheng are you talking about these lines ? 
 <ImportGroup Label="ExtensionSettings">
   <Import Project="$(VCTargetsPath)\BuildCustomizations\CUDA 8.0.props" />
 </ImportGroup>

WuYingzheng, 2018-06-20:
@nicole-antoine no, I'm talking about CUDA setting in you MS-VS . Right click you project ,select property, then you will see the cuda settings. sorry, I am currently working in ubuntu os, can't show you a screen shot of vs.

nicole-antoine, 2018-06-20:
@WuYingzheng I see what you mean though there's no many parameters I can modify like command line or code generation but I think it's unlikely to work, did it do the job for you ?

WuYingzheng, 2018-06-20:
@nicole-antoine  NO. My GPU is GeForce GTX 650, and cuda parameter is  compute_30 sm_30.

nicole-antoine, 2018-06-23:
@WuYingzheng Alright I changed compute_/sm_(35) to 30 and it did the trick thank you.  Now I'm at the same step as @quanqhow  was, it gets stuck after several frames. I tried as @junweiy2012 suggested by commenting "RUN_MULTITHREADED", the program isn't stuck anymore but the window becomes black after displaying the same number of frame.

yasinalm, 2018-07-13:
I solved the "out of memory" error with changing parameters s_hashNumBuckets  and s_hashNumSDFBlocks with smaller numbers. You can try this. My GPU is gtx 1050ti.

lmlodda, 2018-07-31:
i have a solutions that works for me with RUN_MULTITHREADED
In DepthSensing.cpp line 987 replace
**bGotDepth = g_CudaImageManager->process()**
with
**bool bGotDepth; 
	while (!(bGotDepth = g_CudaImageManager->process()));**

The problem that i investigated was that the kinect one provides 30fps max. If the frame is not ready to fetch, pDepthFrameReference->AcquireFrame returns a FAIL. This causes a dead lock in the process lock management. The while loops forces the CALLBACK to wait untill a new frame is available. Hope this helps.

farray81, 2019-02-27:
@lmlodda Which version of CUDA, Visual Studio, Windows did you achieve that Kinect v2 work with? I am trying some things said in this post, included your suggestion, and the program doesn't work.

tansangxtt, 2019-02-28:
> i have a solutions that works for me with RUN_MULTITHREADED
> In DepthSensing.cpp line 987 replace
> **bGotDepth = g_CudaImageManager->process()**
> with
> **bool bGotDepth; while (!(bGotDepth = [...]

Wonderful @lmlodda, I will try your solution

farray81, 2019-02-28:
@tansangxtt Did you achieve it worked? I tried the solution purposed by @lmlodda and it didn't work for me.

rapulu92, 2019-05-29:
@yaoy908 
I modify the sensfilereader class, to read my images from kinect, so I didn't try it with realtime-running.On my pc, kinectonesensor cannot work well. You can modify it, but my pc cannot affort a good fps with realtime-running, so I choose to read images from disk.

Please how did you modify it?

farray81, 2019-09-30:
@wuyingnan
Do you know (or anyone knows) if this modern port for BundleFusion solves the problems with Kinect One?

https://github.com/wuyingnan/BundleFusion

rapulu92, 2019-10-02:
yes i did use kinect one

https://github.com/wuyingnan/BundleFusion

https://github.com/niessner/BundleFusion/issues/6#issuecomment-536502554
-------------------------------------------------- 
AUGUSTUS RAPULUCHUKWU OKOYENTA 
Graduate Student, School of Mechanical Engineering, 
Shanghai Jiao Tong University, China. 
M: 18621106438.

farray81, 2019-10-02:
Oh, great! As you see in the post, a lot of people were trying to use BundleFusion with the second version of the Kinect, the Kinect One, and we didn't achieved it. I hope I can try your version in the next days.

carstenschwede, 2019-10-02:
@farray81 Please report back if https://github.com/wuyingnan/BundleFusion works with Kinect One. Looking forward to your results!

farray81, 2019-10-04:
Unfortunately, it doesn't work in my case. I achieved to try this version (https://github.com/wuyingnan/BundleFusion) with VS2017, CUDA10.1 and Windows 8, but the same problem than before appeared. Black screen stuck, and if I change the code as @Imlodda said:

> i have a solutions that works for me with RUN_MULTITHREADED
> In DepthSensing.cpp line 987 replace
> **bGotDepth = g_CudaImageManager->process()**
> with
> **bool bGotDepth; while (!(bGotDepth = g_CudaImageManager->process()));**

then the screen is not stuck, but when the program read 10 frames, the message INVALID CHUNK appears and the program finishes.

I don't know what happens, because I can use the Kinect One with https://github.com/niessner/VoxelHashing with no problems, that it's the previous project to BundleFusion.  I'm always trying the Kinect in real-time, I don't know how to do to use the data from this sensor offline in BundleFusion, as @yaoy908 said;

> I modify the sensfilereader class, to read my images from kinect, so I didn't try it with realtime- running.On my pc, kinectonesensor cannot work well. You can modify it, but my pc cannot affort a good fps with realtime-running, so I choose to read images from disk.

wuyingnan, 2019-10-24:
I do not have a Kinect One sensor. This sensor is not tested. The code working with Kinect One is the same as original bundle fusion

FishHe, 2020-05-15:
Same Problem I have met. Is there any solution?

FishHe, 2020-05-16:
I follow the program, it stop at SiftPyramid.cpp line 445:
 ProgramCU::ComputeOrientation(ftex, got, key, sigma, sigma_step);
I don't know what happened.. Hope for a solution.

FishHe, 2020-05-16:
I can't compile your code with VS 2017, could you tell me how to do it? @wuyingnan

chethanab16, 2020-08-18:
Hi I have this issue when i run the bundle fusion and i am using windows 10, visual studio 2013, and tried with CUDA 8 , 9 , and 11 version but  has the same problem in the imaged could any body suggest how to solve it

[image]

BruceWANGDi, 2020-10-12:
@chethanab16  Hi, I have the same problem with you, my environment is the same as yours(And my hardware is 2080Ti, I don't know whether my GPU can not work on CUDA 8.0). Have you figured out any solutions to solve it?
When I running my .exe, it stuck, then crash without any response.
Looking forward to your reply so much.

chethanab16, 2020-10-12:
@BruceWANGDi try  1. https://github.com/niessner/BundleFusion/pull/56/files

                                  2.https://github.com/niessner/BundleFusion/issues/6
this should solve the GPU Hang Issue and I  tried with CUDA 10.2

BruceWANGDi, 2020-10-12:
@chethanab16  Thank you so much!!! I changed my CUDA version8.0 to 10.0 and followed the #56 you give me. I can run the project finally!!
Overall, I think the most important thing is that my previous CUDA is too old so that my GPU can't work on it, and I think it's essential to change the code described by #56 when using 20 series graphic card.

Jason-YJ, 2021-03-01:
@chethanab16
Have you solved the problem by changing cuda version from 8.0 to 10.2?
My pc is just the same situation exactly, even a same GPU -1660Ti

crud89, 2022-04-01:
For anyone still interested, I've got this running. A fixed version can be found [here](https://github.com/crud89/BundleFusion/tree/kinect-one). I've also integrated the fixes #28, #31 and #56 by @stotko. Unfortunately, it currently only works in single-threaded mode. If anybody wants to provide a fix for this, be my guest, but I find the single-threaded performance more than sufficient.

The fork works on CUDA 11.5, but I also provide a description how to adjust the CUDA version. I suppose all versions newer than CUDA 7 will work.

I also wanted to integrate the changes by @wuyingnan (they are great!), but unfortunately, `mLib` does define a hard dependency on the legacy D3DX11 shader compiler, which requires manual patching, so I opted to rely on the DX SDK for simplicity reasons.

Furthermore I've updated the `mLib` dependency up to the last commit that is compatible with the externals they provide. More recent versions require you to manually update boost in the externals, so if you want to use it, you would have to keep this in mind!

[image]

tansangxtt, 2022-04-10:
Thanks @crud89 for your work, I will try it as soon as I get a Kinect XBox One

MisEty, 2023-04-18:
> For anyone still interested, I've got this running. A fixed version can be found [here](https://github.com/crud89/BundleFusion/tree/kinect-one). I've also integrated the fixes #28, #31 and #56 by [...]

I have test your code and test on two laptops. The building process is OK but only the debug build can work. The release type crash on some frames and I don't know why. Anyway, Thanks for your work!