niessner/BundleFusion #7: Is it possible to convert to Linux platform?

daleydeng, 2017-08-18:
If it is, could you provide some roadmap or hints?

rnunziata, 2017-09-23:
I would also like to see this.....if you cold just isolate  the windows calls so a different framework can be plugged in.

alexsmartens, 2018-04-01:
+1 I'm also interested

nonlinear1, 2019-08-07:
I have finished porting bundlefusion to linux platform except display, because I don't understand directx!. My bundlefuison for linux can run any dataset of bundlefuison. I think the result is good.
I want to upload github. but I failed because the chunk size of file is constrained to 50M.

TheodoreT, 2019-11-21:
@nonlinear1 is it possible to find an alternative way to upload the files somewhere? What kind of file exceeds the 50Mb limit?

Regarding the directX thing, there is another more recent version [here](https://github.com/wuyingnan/BundleFusion) which does not use directX. Maybe this could solve the visualization part that you are mentioning.

nonlinear1, 2019-11-25:
I have upload my linux verison of bundlefusion: https://github.com/nonlinear1/BundleFusion_Ubuntu_V0

mrlittlepig, 2019-12-02:
@nonlinear1 Can you plugin you mLib code to this ubuntu repo? I have get some problem while compile BundleFusion_Ubuntu_V0 with mLib, thank you

nonlinear1, 2019-12-02:
I try to pull the mLib to this ubuntu repo. but l failed. If you want the mlib, changed by me, I will send you by email

mrlittlepig, 2019-12-02:
@nonlinear1 my email is: mrlittlezhu@gmail.com, thank you~

nonlinear1, 2019-12-02:
I try to send you mLib, but I failed,

mrlittlezhu@gmail.com | 被对方服务器退回
-- | --

TheodoreT, 2019-12-02:
@nonlinear1 what is the issue that you have uploading the mLib at your repo? We should solve it somehow or find an alternative since in practice by mailing it individually to people is a kind of impractical.

nonlinear1, 2019-12-02:
another email? @mrlittlepig

mrlittlepig, 2019-12-02:
@nonlinear1 Or use this email: zhangjunhui@cvte.com

nonlinear1, 2019-12-02:
@TheodoreT I cann't upload the files, which is in a folder! I guess maybe my environment is something wrong!

TheodoreT, 2019-12-02:
@nonlinear1 can you upload the mLib folder to the https://wetransfer.com/ and then share the link (there are two options either use a mail directly or upload the files and then share the link) then I could try to make a pull request and see whether this works or not.

vividaner, 2019-12-03:
@nonlinear1 Could you please send the complete ubuntu version to my email, miaodaner@gmail.com? Thanks a lot!

nonlinear1, 2019-12-04:
the file is two large about 4 G，also I cann't send any email to gmail.

nonlinear1, 2019-12-04:
@TheodoreT for me, I upload the file is too slow ,but I will try! If I succeed, I will tell you!

nonlinear1, 2019-12-04:
TheodoreT I wil try

Arnoldnuo, 2019-12-10:
> TheodoreT I wil try, In all, my project about 4G

Can you build a docker image to run this project, so we can run bundlefusion via docker.

nonlinear1, 2019-12-26:
Now, I have solve the issue for upload my BundleFusion_Ubuntu, Anyone can download from
https://github.com/nonlinear1/BundleFusion_Ubuntu_V0 Don't forget give me a star :)