niessner/BundleFusion #8: how to install the package and how to run it

amiltonwong, 2017-08-22:
Hi, authors,

Thanks for releasing BundleFusion. Could you also list briefly how to install the package and demo steps?

Thanks!

adammpolak, 2018-01-19:
https://bericht.neopostmodern.com/posts/artist-guide-to-bundlefusion}

Some one already hooked it up ^ :)

cyummy, 2018-03-28:
I can't get access into the URL you mentioned. @adammpolak

alexsmartens, 2018-04-01:
The same problem here. Is there any documentation for this project?

BTW, @adammpolak your link does not work. Do you have a different link?

bruceche11, 2018-05-25:
adammpolak's link is right, just delete the last }

sd707589, 2019-10-16:
https://bericht.neopostmodern.com/posts/artist-guide-to-bundlefusion

nirmalsnair, 2024-01-14:
Try this: https://welcometomywebsite.neopostmodern.com/posts/artist-guide-to-bundlefusion