niessner/BundleFusion #10: core-base/common.h cannot be found.

HyperLethalVector, 2017-08-25:
Hi, I seem to be missing a file after downloading the source and accompanying mLib external.

Error	45	error C1083: Cannot open include file: 'core-base/common.h': No such file or directory	d:\bundlefusion-master\friedliver\source\mLibCuda.h	2	1	FriedLiver

is there something I'm still missing? I'm using Cuda 7.0 to compile.

neopostmodern, 2017-08-30:
Not sure if related, but maybe it helps.

I get the exact same error (VS2013 Community with Update 5, Windows 10 without Anniversary Update, CUDA 7.0 but with recent driver) and the next error is:
```
Error	2	error MSB3721: The command ""C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v7.0\bin\nvcc.exe" -gencode=arch=compute_35,code=\"sm_35,compute_35\" --use-local-env --cl-version 2013 -ccbin "C:\Program Files (x86)\Microsoft Visual Studio 12.0\VC\bin\x86_amd64" -rdc=true -IInclude\cutil\inc -I"C:\Users\C\Downloads\BundleFusion-master-1\BundleFusion-master\FriedLiver" -ISource\DXUT\Optional -ISource\DXUT\Core -IInclude\Uplink -I"C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v7.0\include" -I"C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v7.0\include"     --keep-dir x64\Debug -maxrregcount=62  --machine 64 --compile -cudart static  -g   -DWIN32 -D_DEBUG -DDEBUG -DPROFILE -D_CONSOLE -DD3DXFX_LARGEADDRESS_HANDLE -D_CRT_SECURE_NO_WARNINGS -DNOMINMAX -D_UNICODE -DUNICODE -Xcompiler "/EHsc /W3 /nologo /Od /Zi /RTC1 /MDd  " -o x64\Debug\CUDACache.cu.obj "C:\Users\C\Downloads\BundleFusion-master-1\BundleFusion-master\FriedLiver\Source\CUDACache.cu"" exited with code 2.	C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V120\BuildCustomizations\CUDA 7.0.targets	593	9	FriedLiver
```

neopostmodern, 2017-08-30:
So, after a lot of digging around I think this is because it doesn't find the `mLib` submodule.
Check that you're code has this structure:
```txt
BundleFusion-master/
  external/
    mLib/ # this is the submodule you replaced
      data/
      src/
      [...]
  FriedLiver/
    [...]
    FriedLiver.sln
    [...]
mLibExternal/ # you downloaded this from Dropbox
  include
  libsWindows
  [...]
```

(or have a look at this run-through I am trying to document: https://bericht.neopostmodern.com/posts/artist-guide-to-bundlefusion)

HyperLethalVector, 2017-08-30:
Hi,

it seems I was missing the entire mLib section, 
when I pulled from bundlefusions repository, it didn't pull the additional mLib git repository.

Setting the folder structure up the way you described seemed to do the trick for compiling.

Running I get the following error:
The code execution cannot proceed because FreeImage.dll was not found. Reinstalling the program may fix this problem.

HyperLethalVector, 2017-08-30:
Nevermind, after reading your post (Thanks for that by the way!)

I successfully ran the program (Needed to switch to release).

I will test it tomorrow with a depth camera and see how we go! :D

closed 2017-08-31 by HyperLethalVector

VVingerfly, 2017-08-31:
@HyperLethalVector Hi, do you run the code with Kinect One and perform real-time reconstruction successfully? When I run the code with VS2015, it just crashed. 

What's your computer configurations when running this code?

Thanks you!

Best Regards,
Wei

happydj94xk4, 2017-09-06:
@neopostmodern and @HyperLethalVector 
Hi both, I ran into MSB3721 problem the same as described by @neopostmodern, but can't access https://bericht.neopostmodern.com/posts/artist-guide-to-bundlefusion to get the solution.
May you describe the solution briefly?

Thanks,
Meng

neopostmodern, 2017-09-07:
@happydj94xk4 — sorry, server outage. It's online again now.

You need to set up the directories correctly, see the previous comment of mine or the post (which is more detailed).

Jaiy, 2017-12-01:
@neopostmodern 
Hi, I have read your document, it's useful for me, thanks a lot.  I have built the solution successfully, but when I try to run it, it failed as follow.  The command window stucked after running, in the command it shows the error——cudaSafeCall() Runtime API error 74: misaligned address. I'm using CUDA 6.5, VS2013，Win8.

neopostmodern, 2017-12-05:
I think they advise against anything other than CUDA 7. Most likely your 6.5 is the issue, but I'm guessing here, so don't nail me down on that. But it's the first thing I'd try.

yaoy908, 2017-12-06:
I use cuda 8.0, and i think it is not cuda's problem. Kinect api is more
related.

Jaiy, 2017-12-07:
@neopostmodern 
Hi, thanks for your kind help, I have solve the problem by changing the CUDA version, it's strange, but it did work. Thanks again!