niessner/BundleFusion #11: How to run the code with .sens file?

zhanlv600, 2017-09-01:
Hello, I compiled your code and it seemed can't run without the depth camera hard device. I found you provided the .sens file, so I wonder if the code can run with the .sens file? If it does, then how to run it with the .sens file?

zhanlv600, 2017-09-01:
I get the answer. Edit the file zParametersDefault.txt: s_sensorIdx = **8**, and copy the .sens file in the project, use 3 arguments, example: zParametersDefault.txt zParametersBundlingDefault.txt test.sens, finally the program will run.

closed 2017-09-01 by zhanlv600

kuonangzhe, 2017-10-18:
@zhanlv600 
hi, did you got the colored reconstruction result? I made it run but the window 'Fried Liver' showing is dark and the output results are also blank images(dark). Thank you so much!