niessner/BundleFusion #16: error C2015: too many characters in constant

guangquanwang, 2017-12-21:
Hi, thanks for this wonderful work. Rescently, I have pull down the latest codes and built it with VS2013 && CUDA7.5. However some errors occurs as follows:
    C:\ThirdPartySoftWare\vs2013\VC\include\concrt.h(5658): error C2015: too many characters in constant
1>C:\ThirdPartySoftWare\vs2013\VC\include\concrt.h(5658): error C2146: syntax error : missing ')' before identifier 'q'
1>C:\ThirdPartySoftWare\vs2013\VC\include\concrt.h(5658): error C2001: newline in constant
1>C:\ThirdPartySoftWare\vs2013\VC\include\concrt.h(5671): error C2143: syntax error : missing ';' before '{'
1>C:\ThirdPartySoftWare\vs2013\VC\include\concrt.h(5671): error C2143: syntax error : missing ')' before ';'
1>C:\ThirdPartySoftWare\vs2013\VC\include\concrt.h(5671): error C2447: '{' : missing function header (old-style formal list?)
1>C:\ThirdPartySoftWare\vs2013\VC\include\concrt.h(5671): error C2015: too many characters in constant
1>C:\ThirdPartySoftWare\vs2013\VC\include\concrt.h(5679): error C2015: too many characters in constant
1>C:\ThirdPartySoftWare\vs2013\VC\include\concrt.h(5679): error C2059: syntax error : ')'
1>C:\ThirdPartySoftWare\vs2013\VC\include\concrt.h(5679): error C2143: syntax error : missing ';' before '}'
1>C:\ThirdPartySoftWare\vs2013\VC\include\concrt.h(5689): error C2015: too many characters in constant
1>C:\ThirdPartySoftWare\vs2013\VC\include\concrt.h(5689): error C2018: unknown character '0x60'
1>C:\ThirdPartySoftWare\vs2013\VC\include\concrt.h(5689): error C2143: syntax error : missing ')' before 'constant'
1>C:\ThirdPartySoftWare\vs2013\VC\include\concrt.h(5689): error C2017: illegal escape sequence
1>C:\ThirdPartySoftWare\vs2013\VC\include\concrt.h(5689): error C2001: newline in constant
1>C:\ThirdPartySoftWare\vs2013\VC\include\concrt.h(5696): error C2143: syntax error : missing ';' before '{'
1>C:\ThirdPartySoftWare\vs2013\VC\include\concrt.h(5696): error C2143: syntax error : missing ')' before ';'
1>C:\ThirdPartySoftWare\vs2013\VC\include\concrt.h(5696): error C2447: '{' : missing function header (old-style formal list?)
1>C:\ThirdPartySoftWare\vs2013\VC\include\concrt.h(5696): error C2015: too many characters in constant
1>C:\ThirdPartySoftWare\vs2013\VC\include\concrt.h(5707): error C2015: too many characters in constant
1>C:\ThirdPartySoftWare\vs2013\VC\include\concrt.h(5707): error C2059: syntax error : ')'
1>C:\ThirdPartySoftWare\vs2013\VC\include\concrt.h(5707): error C2017: illegal escape sequence
1>C:\ThirdPartySoftWare\vs2013\VC\include\concrt.h(5707): error C2001: newline in constant
1>C:\ThirdPartySoftWare\vs2013\VC\include\concrt.h(5713): error C2143: syntax error : missing ';' before '{'
1>C:\ThirdPartySoftWare\vs2013\VC\include\concrt.h(5713): error C2143: syntax error : missing ')' before ';'
1>C:\ThirdPartySoftWare\vs2013\VC\include\concrt.h(5713): error C2447: '{' : missing function header (old-style formal list?)
1>C:\ThirdPartySoftWare\vs2013\VC\include\concrt.h(5713): error C2015: too many characters in constant
1>C:\ThirdPartySoftWare\vs2013\VC\include\concrt.h(5713): error C2017: illegal escape sequence
1>C:\ThirdPartySoftWare\vs2013\VC\include\concrt.h(5713): error C2001: newline in constant
1>C:\ThirdPartySoftWare\vs2013\VC\include\concrt.h(5724): error C2015: too many characters in constant
1>C:\ThirdPartySoftWare\vs2013\VC\include\concrt.h(5724): error C2017: illegal escape sequence
1>C:\ThirdPartySoftWare\vs2013\VC\include\concrt.h(5734): error C2015: too many characters in constant
1>C:\ThirdPartySoftWare\vs2013\VC\include\concrt.h(5734): error C2017: illegal escape sequence
1>C:\ThirdPartySoftWare\vs2013\VC\include\concrt.h(5742): error C2015: too many characters in constant
1>C:\ThirdPartySoftWare\vs2013\VC\include\concrt.h(5742): error C2017: illegal escape sequence
1>C:\ThirdPartySoftWare\vs2013\VC\include\concrt.h(5742): error C2001: newline in constant
1>C:\ThirdPartySoftWare\vs2013\VC\include\concrt.h(5750): error C2015: too many characters in constant
1>C:\ThirdPartySoftWare\vs2013\VC\include\concrt.h(5750): error C2001: newline in constant
1>C:\ThirdPartySoftWare\vs2013\VC\include\concrt.h(5758): error C2015: too many characters in constant
1>C:\ThirdPartySoftWare\vs2013\VC\include\concrt.h(5764): error C2015: too many characters in constant
1>C:\ThirdPartySoftWare\vs2013\VC\include\concrt.h(5764): error C2017: illegal escape sequence
1>CUDACache.cudafe1.stub.c(4): fatal error C1004: unexpected end-of-file found
1>  # --error 0x2 --
1>C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V120\BuildCustomizations\CUDA 7.5.targets(604,9): error MSB3721: The command ""C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v7.5\bin\nvcc.exe" -gencode=arch=compute_35,code=\"sm_35,compute_35\" --use-local-env --cl-version 2013 -ccbin "C:\ThirdPartySoftWare\vs2013\VC\bin\x86_amd64" -rdc=true -IInclude\cutil\inc -I"C:\Users\Administrator\Documents\Visual Studio 2013\Projects\BundleFusion-master\FriedLiver" -ISource\DXUT\Optional -ISource\DXUT\Core -IInclude\Uplink -I"C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v7.5\include" -I"C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v7.5\include"     --keep-dir x64\Release -maxrregcount=62  --machine 64 --compile -cudart static --verbose --keep    -use_fast_math -DWIN32 -DNDEBUG -D_WINDOWS -DD3DXFX_LARGEADDRESS_HANDLE -D_CRT_SECURE_NO_WARNINGS -DNOMINMAX -D_UNICODE -DUNICODE -Xcompiler "/EHsc /W3 /nologo /Ox /Zi  /MT " -o x64\Release\CUDACache.cu.obj "C:\Users\Administrator\Documents\Visual Studio 2013\Projects\BundleFusion-master\FriedLiver\Source\CUDACache.cu"" exited with code 2.
1>Done executing task "CudaCompile" -- FAILED.
1>Done building target "CudaBuild" in project "FriedLiver.vcxproj" -- FAILED.
Any body can help with this ,Thanks.

gitliuyf, 2018-01-09:
I can't use CUDA7.5 either.  Try to use CUDA under 6.5 or over 8.0.

angeladai, 2018-01-11:
From what I've heard from other people, this seems to be an incompatibility with cuda 7 and a chinese language pack, but cuda 8 should work.

closed 2018-01-11 by angeladai

WuYingzheng, 2018-05-05:
cuda 9.1 works well