niessner/BundleFusion #19: VS2017 Cuda 9.1

adammpolak, 2018-01-25:
Not sure if this is the right place to leave this info but I was able to compile in VS2017 with Cuda 9.1

### If you want to run VS2017 with Cuda 9.1

- Make sure to download VS2017 with building for C++ (in modify in the installation screen)
- Install Cuda 9.1 *after* VS2017 has been installed
- Change reference to 7.0 in FriedLiver.vcxproj to 9.1

### If you want to run in debug mode

- You will get a FreeImage.dll error
- FreeImage can be found in the mLibExternal/libsWindows/dll64
- Pop that bad boy in FriedLiver/x64/Debug
- Also put the 2 z .txt files in the Debug folder as well

rayhero, 2018-02-01:
@adammpolak : I try to build Bundlefusion using VS2017 with Cuda 9.1 but fail. I meet two problem but I can only fix one of them. the first is the latest VS2017 compile error. It can be solve by modify the vcxproj settings to add 

<Import Project="$(SolutionDir)\Microsoft.VCToolsVersion.14.11.props" /> 

follow below:
https://stackoverflow.com/questions/47780355/cuda-compiling-error-after-installing-it

and also modify the CUDA settings type host_config.h to set _MSC_VER > 1912 like below:

https://devtalk.nvidia.com/default/topic/1022887/cuda-not-working-with-vs-2017-update-/

After do those procedure, I only have one compile error on IDE:
[image]

the C2280 error: attempting to reference a deleted function in D3D11Canvas3D.h. The D3D11Canvas3D.h is the mLib I download from here:

https://github.com/niessner/mLib/issues

Is I using wrong version mLib cause these error? I have no idea how to fix it...

adammpolak, 2018-02-01:
@rayhero how have you set up your mLib in your hierarchy?

```
BundleFusion-master/
  external/
    mLib/ # this is the submodule you replaced
      data/
      src/
      [...]
  FriedLiver/
    [...]
    FriedLiver.sln
    [...]
mLibExternal/ # you downloaded this from Dropbox
  include
  libsWindows
  [...]
```

Use that ^ from: https://github.com/niessner/BundleFusion/issues/10#issuecomment-325975690

rayhero, 2018-02-01:
@adammpolak Yes, I did. I follow your advise:
https://bericht.neopostmodern.com/posts/artist-guide-to-bundlefusion

But I still stuck in C2280 error...I search related reason show that some old project build by VS2013 will have similar problem in VS 2015/VS2017. How can you success build in VS2017 with CUDA9.1?
I really need you help...

rayhero, 2018-02-01:
Finally I let the build successful by roll back to VS2013 + CUDA8...

guyrap, 2018-02-23:
Hi @adammpolak, I've managed to compile with CUDA 9.1 and VS 2017 as you've described. Thank you!
Nevertheless, when I run the code, with the copyroom.sens data, for example, I'm getting the following runtime error: 
CUDASolverBundling.cpp(135) : cutilcheckmsg() CUTIL DATA error : CUDASolverBundling::CUDASolverBundling : (13) invalid device symbol

Can you assist with that? 
BTW, I've also compiled it successfully with CUDA 7 and VS 2013 and the same error appeared.
Thank you, Guy

adammpolak, 2018-02-23:
Have you edited the z files?

-- 

Adam Polak

Product Manager, Citrusbyte

Phone: 203 536 0962
www.citrusbyte.com <https://citrusbyte.com/>

   - <https://twitter.com/citrusbyte>
   - <https://www.facebook.com/citrusbyte>
   - <https://github.com/citrusbyte>
   - <https://dribbble.com/citrusbyte>
   - <https://www.linkedin.com/company/citrusbyte>

guyrap, 2018-02-24:
Hi Adam, I did updated the z files according to previous posts:
s_sensorIdx = 8;
s_binaryDumpSensorFile = "\path\to\copyroom.sens";

I've also reduced the number of images to decrease memory footprint and adjusted the subMapSize accordingly:
s_maxNumImages = 200; //1200
s_submapSize = 30;

Best,
Guy

adammpolak, 2018-03-09:
@guyrap hey sorry were you able to resolve your issue?

guyrap, 2018-03-09:
Hi @adammpolak, actually I've managed to resolve the issue by changing CUDA code generation flags to compute_30,sm_30 (I have a slightly older GPU). It has been running smoothly since then. Thank you! Guy

94zhaofang, 2018-07-16:
@rayhero , Hello, I also meet C2280 error , and I use mLib to build the bundlefusion project by VS2015 + CUDA9.1. Could you tell me how to rebuild the libraries from mLibExternal ? Thank you !

chethanab16, 2020-09-18:
@adammpolak  1.  I try to compile the bundle fusion project using VS 2017 and got the following error,

[image]
 
2.And also I tried with the new mlib using VS 2017 but got the following error, I have followed your steps to compile the bundle fusion project using VS 2017 ,Could you please give me some suggestion on how to solve this issue

[image]

Thankyou

chethanab16, 2020-09-24:
@rayhero @adammpolak @94zhaofang @guyrap  Hi, Anybody  successfully compiled Bundle Fusion with VS 2017