niessner/BundleFusion #20: FriedLiver stuck

Bryce527, 2018-02-01:
Hi, everyone. I have some problems when using Kinect 2.0 for online Reconstruction. In zParametersDefault.txt I changed s_sensorIdx to 2, and it compiled successfully. My platform is Windows 10、Cuda 9.1、VS2013、Kinect SDK 2.0、Microsoft DirectX SDK (June 2010). 
Everytime I set the program running, it stuck, without any response. Thanks very much for any suggestions!

Bryce527, 2018-02-01:
[image]

xiaomaoo, 2018-02-05:
Try to change the parameter "s_maxNumImages " from 1200 to 400, in the zParametersBundlingDefault.txt

liangxu007, 2018-02-07:
@xiaomaoo I tried but it stuck again. Even change to 100, it stuck again.

Keudeugi, 2018-03-08:
Hope this help for you (or the guys who are struggling with build stage w/ Kinect One), but I'm trying to adapt OpenNI2 patched with libfreenect2 make Kinect One work in this project. Currently, I can read only the first frame of stream, but I think It can be solved after I figure out some data processing methods...

[image]

liangxu007, 2018-03-24:
@Keudeugi Hi, have you figured out the data processing methods now?

quanqhow, 2018-03-24:
I also tested my Kinect V2 (for XBox One) under the FriedLiver.exe, but it still stuck at the moment. So I have tested both kinds of Kinect and the FriedLiver stuck at the two conditions.

Keudeugi, 2018-09-08:
@94zhaofang Since my O/S is Windows 10, as well as I didn't installed any additional drivers for DirectX, I think it doesn't matter for Direct X version. One thing I remember is this project is using somehow lower version than default Windows 10's version, which is 12. (In short, a version around ~ X11 is used for this project)

Keudeugi, 2018-09-08:
@liangxu007 Sorry for late reply, but I gave modifying up since it spent too much time (Because the reason that I don't have enough knowledge in this field still). Please mention me if there's a people who make it work with Kinect One.

hejinying, 2019-03-14:
@Keudeugi hi,when i use this project with vs2015, i met the problem here: error C2280: 'ml::D3D11Canvas2D::ElementBillboard::ElementBillboard(const ml::D3D11Canvas2D::ElementBillboard &)': attempting to reference a deleted function
do you know how to solve this problem ??

tkbala, 2019-06-24:
I have the same issue as @hejinying . Any ideas?

FishHe, 2020-05-15:
Same question. The program is stuck at the very beginning.

kuangqi93, 2020-08-13:
> Same question. The program is stuck at the very beginning.

Have you solved that？

FishHe, 2020-08-14:
I have tested on Win10 +CUDA 10.2 + VS2013 + GTX 2080 Super.
The most important fix is: #56

chethanab16, 2020-08-20:
@liangxu007 Did you solve the problem of the window stuck when the program starts.

BruceWANGDi, 2020-10-12:
Hi, guys. There is some problem with my project when I using KinectOneV2(My environment is Win10, VS2013, CUDA8.0, and RTX2080Ti), I don't know whether my GPU can't support CUDA8.0.
When I running my .exe, it stuck, then crash without any response. The FriedLiver window just gets dark.
Does anyone have some solutions to help me to solve this problem? Thank you so much.

mshmoon, 2021-01-22:
I think the reason for this problem is that the thread is locked while waiting for SIFT feature extraction, which makes it unable to continue execution. However, I don't know what causes the feature extraction failure.

Please see DepthSensing.cpp line 1016 and line 1017.

//wait until the bundling thread is done with: sift extraction, sift matching, and key point filtering
ConditionManager::lockBundlerProcessedInput(ConditionManager::Recon);

mshmoon, 2021-01-22:
After further debugging, I found that the program can execute the code to extract SIFT features, but when the program runs to ProgramCu.cu  line 1424,it can be stuck!
`int ProgramCU::CheckErrorCUDA(const char* location)
{
#if (defined(_DEBUG) || defined(DEBUG))
1424 cudaDeviceSynchronize();
	cudaError_t e = cudaGetLastError();
	if (e)
	{
		if (location) fprintf(stderr, "%s:\t", location);
		fprintf(stderr, "%s\n", cudaGetErrorString(e));
		assert(0);
		return 1;
	}
	else
	{
		return 0;
	}
#else`

Jason-YJ, 2021-03-01:
@mshmoon Hi, do you have further progress on solving the problem? Thanks