niessner/BundleFusion #22: .Sens File

franmts, 2018-02-20:
How to create the .sens file for processing NYU dataset like demonstrated by the author?

Here I could see how to decode .sens, but not how to encode.
https://github.com/ScanNet/ScanNet/tree/master/SensReader

Or is it there another method for reading the original .ppm, .pgm and accelerometer .dump files?

whstarxu54, 2018-08-16:
@franmts Have you figured out how to generate .sens file? I want to use my rgb-d camera to first generate a sequence of frames and then use BundleFusion to do 3d reconstruction.

FengLQy, 2018-12-08:
Hello, do you have a  .sen data set about human body? or how to generate .sens file  
thank you very much

carpdm, 2020-09-18:
@franmts @whstarxu54 @FengLQy Do you guys have any solutions? seems that I meet the same problem, Are there any tools to encode to this format?

zzdengke, 2022-01-21:
Hello, have you figured out how to generate .sens file?