niessner/BundleFusion #23: Welcome to BundleFusion learning group!

electech6, 2018-04-08:
Bundle fusion is quite suitable for RGB-D SLAM learners. However, people came across many problems when ran or learned the code, while the authors were too busy to reply the reported issues. So I create a learning group, you can just use WeChat APP to scan the following binary code and we can exchange and learn together!
[image]

chickenbestlover, 2018-05-23:
This  QR code is expired.

electech6, 2018-05-25:
Already updated.

SRenye, 2018-06-04:
The QR code is expired again now ,could you update it :)

electech6, 2018-06-13:
QR code was updated again. If the code is expired again someday, contact my wechat (electech6) with comment "Bundle Fusion" and I will join you in the group.

chethanab16, 2020-09-15:
@electech6 Hi I try to compile the bundle fusion with visual studio 2017 but got the following error , could some one give me suggestion how to solve this problem.
[image]