niessner/BundleFusion #29: How to get the final TSDF without voxel hash?

zzxmllq, 2018-11-02:
How to get the final TSDF without voxel hash?
    Does anyone have the idea?I need the data to test on 3DMV for Semantic segmentation.
    The 3DMV needs voxelsize at 4.7cm and the default of bundlefusion is 1cm. When I change it the resolution will be so hard to distinguish.How to change the parameters to get 4.7cm data?
    Thanks for helping!

LionelLeee, 2019-04-06:
Hello, do you know how to change it? I need 4mm bundlefusion, need to modify those parameters? I hope you can help me answer.

zzxmllq, 2019-04-07:
@LionelLeee I just use https://github.com/christopherbatty/SDFGen to convert the bundlefusion result obj to 4.7cm

zzxmllq, 2019-04-07:
@LionelLeee I just use https://github.com/christopherbatty/SDFGen to convert the bundlefusion result obj to 4cm，if  you want to change it in bundlefusion,maybe you should change the  parameters in zParametersDefault.txt