niessner/BundleFusion #32: runtime API error 10： invalid device ordinal

calyx1991, 2018-11-30:
[image]
I just set 's_sensorIdx = 8;' in zParametersDefault.txt, but when I run the project, I get the mistake runtime API error 10： invalid device ordinal. My computer only have one GPU, so I think this is the problem, but how to solve it ?

BrigitteBai, 2018-12-11:
I met the same problem. Can I ask what is your graphics card？
Now I suspect that may be caused by insufficient graphics card memory……

BruceWANGDi, 2020-10-20:
@jinzijian Hi, did you figure out any solution to solve this issue? My device is KInectV2.

BruceWANGDi, 2020-10-20:
@BrigitteBai Hi, my GPU is gtx1650, and I met the same issue as yours. Did you figure out any solution to solve that?

BruceWANGDi, 2020-10-21:
@calyx1991 I met the same issue with you, did you solve it? I have no idea about this problem.

sjtuspeechless, 2021-02-21:
my gpu is rtx2080，and I met the same problem. Can anyone help us?