niessner/BundleFusion #34: The output file of apt0.sens seems have many regular cavities

h1063135843, 2018-12-03:
I ran the program  with the default parameter. The input file is apt0.sens.  I viewed the output file in MeshLab, and it has many regular cavities. What's wrong with it?
[image]

FengLQy, 2018-12-06:
can you tell me how to run it by dataset?I don't have a depth camera, but I want to run it. thank you very much

xiaomaoo, 2018-12-12:
How do you get the output file? is it a .ply file? 
the soure project seems not include the method to export the  ply model
Could you tell me please?

h1063135843, 2018-12-12:
> How do you get the output file? is it a .ply file?
> the soure project seems not include the method to export the ply model
> Could you tell me please?

In fact, I ran the program and when it ended the output file appears automatically. It is a .ply file.

xiaomaoo, 2018-12-13:
> > How do you get the output file? is it a .ply file?
> > the soure project seems not include the method to export the ply model
> > Could you tell me please?
> 
> In fact, I ran the program and when it ended the output file appears automatically. It is a .ply file.
Thanks for reply!
Have you modified any values in the two parameters.txt?

h1063135843, 2018-12-13:
> > > How do you get the output file? is it a .ply file?
> > > the soure project seems not include the method to export the ply model
> > > Could you tell me please?
> > 
> > 
> > In fact, I ran the program and when it ended the output file appears automatically. It is a .ply file.
> 
> Thanks for reply!
> Have you modified any values in the two parameters.txt?
> Is it scans/scan.ply?
It's not scan.ply. The name is the same as the input.  I just set the sensorIDX to 8 and change the path of Input file.

h1063135843, 2018-12-13:
Sometimes I ran the kinect and press '7'  to save, the problem also occurs.  Sometimes it works perfectly and I don't know why.

Hegs30, 2019-03-16:
> Thanks for reply!
> Have you modified any values in the two parameters.txt?
> Is it scans/scan.ply?

I can‘t get the output file too，did you finally get it？

zhongtao93, 2020-07-16:
I get the same wrong output ply. Did anyone solve it?

[image]

Thanks~

h1063135843, 2020-07-16:
> I get the same wrong output ply. Did anyone solve it?
> 
> Thanks~

There is a parameter named s_marchingCubesMaxNumTriangles  in zParametersDefault.txt.
In marchingcube phase ,when the triangle number exceeds the threshold, the program will decrease triangle number first.
This is the reason leading to cavities.Try increase the parameter.
However, the max you could try depends on you GPU memory size.

closed 2020-07-16 by h1063135843