niessner/BundleFusion #39: How to generate global .ply Mesh file?

xiaomaoo, 2018-12-12:
When I press the key '9' ,  StopScanningAndExtractIsoSurfaceMC() will be excuted. 
But  it seems only genetate current local map, Why?
How to generate global .ply Mesh file?
[image]

backkon, 2019-04-21:
@xiaomaoo why i press the key '9',it has no reaction?

backkon, 2019-04-21:
@xiaomaoo look forward to your reply.

sd707589, 2019-10-16:
@xiaomaoo @backkon It seems that only when your frames are calculated their poses correctly, those frame can be included in the output mesh. 
But the diverse frames are hard to match each other, especially in the situation of rotation.

@angeladai Can you give any suggestions of how to alter the XXXParameters.txt?

Serena2018, 2020-08-20:
When the projection is excuted successfully, the global ply will be save into the path same with the path of your input .sens file with the same name with your input .sens file but with a ply postfix.