niessner/BundleFusion #40: frame-XXXXXX.pose.txt

zhujiawen1994, 2018-12-27:
Hello,sir,
        I would like to know if I want to get the pose information of camera about each frame , that is, the transformation Ti = Ri p + ti (rotation Ri , translation ti )  in the paper, which files should I refer to?

Tomas-Lee, 2019-10-16:
I want to know t, too

Serena2018, 2020-08-20:
You know that after bundlefusion is operated, frame-XXXXXX.pose.txt wrapped in your .sens file is updated. You can translate the newest .sens file into color ones, depth ones and pose ones, then you can each camera pose information for each frame.

chethanab16, 2020-11-25:
@Serena2018 could you elaborate more about "**translate the newest .sens file into color ones, depth ones and pose ones, then you can each camera pose information for each frame.**  how to do it 

@zhujiawen1994  @Tomas-Lee  if any one know how to get the frame-XXXXXX.pose.txt  please let me me know  how to get it

ph-code-repo, 2022-03-10:
The functionality for it is already in the PoseHelper.h file, So I added a few lines to deinitBundleFusion in BundleFusion.cpp (few lines above and below given so you can find it in the source code):

`#ifdef RUN_MULTITHREADED
    g_bundler->exitBundlingThread();

    g_imageManager->setBundlingFrameRdy();			//release all bundling locks
    g_bundler->confirmProcessedInputFrame();		//release all bundling locks
    ConditionManager::release ( ConditionManager::Recon ); // release bundling locks
    
    //NEW CODE HERE
    std::vector<mat4f> trajectory;
    g_bundler->getTrajectoryManager()->getOptimizedTransforms ( trajectory );
    PoseHelper::saveToPoseFile(GlobalAppState::get().s_generateMeshDir + "poseFile.txt",trajectory);

    if ( bundlingThread->joinable() )
        bundlingThread->join();	//wait for the bundling thread to return;
#endif`

This should be run if you press 'q' to exit and shutdown the visualisation, and output poseFile.txt in the directory where a mesh would be saved, as specified by s_generateMeshDir in zParametersDefault.txt
Some of the code is copied from processInputRGBDFrame in BundlFusion.cpp, where there is a section for publish_mesh to the visualisation - it gets the trajectory, splits it into positions and camera poses to draw to the visualisation. I just sent the trajectories before splitting into the PoseHelper::saveToPoseFile function. Some of the values can be expressed in scientific notation.