niessner/BundleFusion #43: Cannot run VS2015 with Cuda 10.0

OllieLee1999, 2019-07-12:
Hi,guys! I tried to run BundleFusion in vs2015,but failed because this error happened in my IDE ：
Compiler Error C2280 “attempting to reference a deleted function” .
After a survey on the Google I found the diffience between VS2013 and VS2015.
See this：
(https://stackoverflow.com/questions/31264984/c-compiler-error-c2280-attempting-to-reference-a-deleted-function-in-visual)
And this:
(https://stackoverflow.com/questions/30120573/visual-studio-2013-does-not-delete-the-copy-constructor-when-a-user-defined-move)
And this:
https://docs.microsoft.com/en-us/cpp/cpp/explicitly-defaulted-and-deleted-functions?view=vs-2015
When I tried to force the copy constructor = default,it didn't work.And there were lots of error C2280.
Any suggestion for this error?
Best,
Guy

FishHe, 2020-05-16:
'auto _e = e.castBillboard();'
The **auto** has some bugs. Changing the code into
'const ElementBillboard& _e = e.castBillboard();' 
can solve the problem. 
@hejinying @rayhero @94zhaofang

Junkai-Liu, 2020-12-08:
@FishHe Hey,this method only can solve part of problem, But there still many errors C2280.By the way ,How to fix move constructor problem.For example,
D3D11Canvas2D(D3D11Canvas2D&& other) {
			m_graphics = nullptr;
			swap(*this, other);     // dont match the swap fuction definition
}
you know, the move constructor input coefficient is right value, but the swap function definite the input coefficient is left value.This lead to a dont match error, how to fix this problem?

Junkai-Liu, 2020-12-10:
@Pengxiang-Li  Hi, have u solved this problem?