niessner/BundleFusion #48: Please help this none coder run this program

amorell1234, 2019-09-28:
I know nothing about coding. I own a Kinect and want to do room scanning.

I installed Visual Studio but how do you get this up and running scanning a room? When I press the sln file I dont know what to do and if I press debug or build it freezes.
I have the mlabExternal folder in there too.

Can somebody write a step by step please.

tim-s7, 2019-10-23:
I think this guide will be useful:

https://bericht.neopostmodern.com/posts/artist-guide-to-bundlefusion

tim-s7, 2019-10-23:
#8

Tomas-Lee, 2019-10-24:
https://blog.csdn.net/ffcjjhv/article/details/89221992
http://www.pianshen.com/article/2929189864/
will also be helpful

nirmalsnair, 2024-01-15:
> I think this guide will be useful:
> 
> https://bericht.neopostmodern.com/posts/artist-guide-to-bundlefusion
> 

Updated link: https://welcometomywebsite.neopostmodern.com/posts/artist-guide-to-bundlefusion