niessner/BundleFusion #49: bundle fusion automatically exit when setting s_SDFVoxelSize=0.004f

Tomas-Lee, 2019-10-27:
In the test Dataset office3.sens, when limiting the  voxel size into 4mm, the program will stuck after outputting "<<<[Frame: 222] 273 >>>" , then it exits without any more output.
But when setting the voxelsize more than 7mm, the program will run OK.
I check out the GPU memory comsumption and CPU memory comsumption, but there are still a lot free space. 
My hardware configuration is win10+cuda8.0+vs2013+GTX1080ti
the parameter I changed is only the following:
s_SDFVoxelSize = 0.004f;

Does anyone know the reason?

NULLcy, 2019-11-10:
i don't know

yan-zeng-quan, 2021-08-03:
Increasing the value of 's_marchingCubesMaxNumTriangles' in txt file ‘zParametersDefault.txt’ could be useful.

18721258653, 2021-11-01:
@Tomas-Lee Increasing the value of 's_hashNumBuckets' and 's_hashNumSDFBlocks' in txt file ‘zParametersDefault.txt’ maybe useful.

NULLcy, 2021-11-04:
OK，thank you 

发自我的iPhone