niessner/BundleFusion #50: Why occured the code ?

CaffeMrDe, 2019-12-17:
I try to run the program. But It usually occured the problem. It flowering : I has get the image successfully ,but it show **INVALID FIRST CHUNK** and exit automatily.

```
PS D:\3DRe\BundleFusion\FriedLiver\x64\Release> .\FriedLiver.exe
usage: DepthSensing [fileNameDescGlobalApp] [fileNameDescGlobalTracking]
fileNameDescGlobalApp = D:/3DRe/BundleFusion/FriedLiver/x64/Release/zParametersDefault.txt
fileNameDescGlobalBundling = D:/3DRe/BundleFusion/FriedLiver/x64/Release/zParametersBundlingDefault.txt

s_RenderMode = 1
s_SDFIntegrationWeightMax = 99999999
s_SDFIntegrationWeightSample = 1
s_SDFMarchingCubeThreshFactor = 10.0f
s_SDFMaxIntegrationDistance = 3.0f
s_SDFRayIncrementFactor = 0.8f
s_SDFRayThresDistFactor = 50.0f
s_SDFRayThresSampleDistFactor = 50.5f
s_SDFTruncation = 0.06f
s_SDFTruncationScale = 0.02f
s_SDFUseGradients = false
s_SDFVoxelSize = 0.010f
s_bUseCameraCalibration = false
s_binaryDumpSensorFile = ../data/sequence.sens
s_binaryDumpSensorUseTrajectory = false
s_colorFilter = false
s_colorSigmaD = 2.0f
s_colorSigmaR = 0.1f
s_depthFilter = false
s_depthSigmaD = 2.0f
s_depthSigmaR = 0.1f
s_garbageCollectionEnabled = true
s_garbageCollectionStarve = 0
s_generateVideo = false
s_generateVideoDir = output/
s_hashMaxCollisionLinkedListSize = 7
s_hashNumBuckets = 800000
s_hashNumSDFBlocks = 200000
s_integrationEnabled = true
s_integrationHeight = 240
s_integrationWidth = 320
s_lightAmbient = 0.4f 0.4f 0.4f 1.0f
s_lightDiffuse = 0.6f 0.52944f 0.4566f 0.6f
s_lightDirection = 0.0f -1.0f 2.0f
s_lightSpecular = 0.3f 0.3f 0.3f 1.0f
s_marchingCubesMaxNumTriangles = 3000000
s_materialAmbient = 0.75f 0.65f 0.5f 1.0f
s_materialDiffuse = 1.0f 0.9f 0.7f 1.0f
s_materialShininess = 16.0f
s_materialSpecular = 1.0f 1.0f 1.0f 1.0f
s_maxFrameFixes = 10
s_minPoseDistSqrt = 0.0f
s_numSolveFramesBeforeExit = 30
s_playData = true
s_printConvergenceFile =
s_printTimingsDirectory =
s_rayCastHeight = 240
s_rayCastWidth = 320
s_reconstructionEnabled = true
s_recordCompression = true
s_recordData = false
s_recordDataFile = dump/recording.sens
s_recordDataHeight = 480
s_recordDataWidth = 640
s_remappingDepthDiscontinuityThresLin = 0.01f
s_remappingDepthDiscontinuityThresOffset = 0.012f
s_renderDepthMax = 4.0f
s_renderDepthMin = 0.1f
s_renderingDepthDiscontinuityThresLin = 0.001f
s_renderingDepthDiscontinuityThresOffset = 0.012f
s_sensorDepthMax = 4.0f
s_sensorDepthMin = 0.1f
[... 178 lines cut ...]
#INFO get image successfully ...
<<< [Frame: 2 ] 200000 >>>
#INFO get image successfully ...
<<< [Frame: 3 ] 200000 >>>
#INFO get image successfully ...
<<< [Frame: 4 ] 200000 >>>
#INFO get image successfully ...
<<< [Frame: 5 ] 200000 >>>
#INFO get image successfully ...
<<< [Frame: 6 ] 200000 >>>
#INFO get image successfully ...
<<< [Frame: 7 ] 200000 >>>
#INFO get image successfully ...
<<< [Frame: 8 ] 200000 >>>
#INFO get image successfully ...
<<< [Frame: 9 ] 200000 >>>
#INFO get image successfully ...
**INVALID FIRST CHUNK**
<<< [Frame: 10 ] 200000 >>>

**OnD3D11FrameRender StopScanningAndExit(true)**
**[ stop scanning and exit ]**
```

Thank you for your reply!

jamoro24, 2020-01-27:
Try setting your camera elsewhere. I encountered the same problem but it solved by change the camera first view.

CaffeMrDe, 2020-03-17:
Yes. I have the set the camera param again. Then it works!

closed 2020-03-17 by CaffeMrDe

jhljx, 2020-12-17:
So how to set the camera parameter? I also encountered the same problem.