niessner/BundleFusion #51: rtx2080ti Runtime API error 73

wangniancai, 2020-03-07:
I try to run bundlefusion with nvidia rtx2080 ti ,cude error ,Source\SiftGPU\SiftPyramid.cpp(387) : cudaSafeCall() Runtime API error 73: an illegal instruction was encountered.
1080& 1080ti worked

276385109, 2020-05-08:
I try to run bundlefusion with nvidia rtx2080S  and the ProgramCU.cu file didn't work.      Have you solved your problem？

FishHe, 2020-05-15:
Same Question. 
When I use compute_61, sm_61 on VS2013 CUDA 10.2, it seems working. 
The NVIDIA says GTX2080 should use sm_75. However there is no implementation in CUDA 10.2 on Windows yet. SM_61 can work indeed.

FishHe, 2020-05-15:
The code can compile, however can not work on 2080.

FishHe, 2020-05-16:
When I use GTX2080, the program stops at 
`unsigned int ProgramCU::ReshapeFeatureList(CuTexImage* raw, CuTexImage* out, int* d_featureCount, float keyLocScale)`
and i use printf to test `ReshapeFeatureList`:
```
	unsigned int res;
        printf("Test begin!");
	cudaMemcpy(&res, d_featureCount, sizeof(int), cudaMemcpyDeviceToHost);
        printf("Test pass!");
```
cmd will only print "Test begin!", and will stops at cudaMemcpy().
I don't know how to solve the bug.

stotko, 2020-05-29:
I am also observing the mentioned GPU hangs. I submitted a respective pull request (see #56) to fix this issue. Until it gets eventually merged, you can apply and try the changes locally on your machines.

FishHe, 2020-06-01:
Thanks! @stotko 
It (https://github.com/niessner/BundleFusion/pull/56) really works!

zhoushengdi, 2021-12-30:
@wangniancai  Hi,I came across the same error, and I have fixed it in my case. This error is from the CMakeLists configuration, cause I run it in RTX 2070, but the default config for"nvcc -arch " is sm_61, which corresponds to the GTX 1080, so I just modify the "set(CUDA_NVCC_FLAGS **-arch=sm_61**;-03;-G;-g;-std=c++11)" to "set(CUDA_NVCC_FLAGS **-arch=sm_75**;-03;-G;-g;-std=c++11)" in CMakeLists.txt(around row 55). And finally, it worked.
> I try to run bundlefusion with nvidia rtx2080 ti ,cude error ,Source\SiftGPU\SiftPyramid.cpp(387) : cudaSafeCall() Runtime API error 73: an illegal instruction was encountered. 1080& 1080ti worked

busybeaver42, 2022-07-13:
Running on Notebook with RTX 2070 does't work.
I have check #56. and it is included and I have changed the CMakeList.txt file:
CUDA_NVCC_FLAGS -arch=sm_75;-03;-G;-g;-std=c++11)
I try to run bundlefusion with nvidia rtx2080 ,cude error ,Source\SiftGPU\SiftPyramid.cpp(387) : cudaSafeCall() Runtime

Z-Jeff, 2022-08-09:
> Running on notebook with RTX 2070 doesn't work. I have check #56. and it is included and I have changed the CMakeList.txt file: CUDA_NVCC_FLAGS -arch=sm_75;-03;-G;-g;-std=c++11) I try to run bundlefusion with nvidia rtx2070 ,cude error ,Source\SiftGPU\SiftPyramid.cpp(387) : cudaSafeCall() Runtime
> 
> Notebook CPU Kerne GHZ max GHz Display SCHENKER XMG PRO 15 – M19gfp Gen9 i7-9750H 6/12 2,60 GHz 4,50 GHz 15,6",FHD, 144 Hz, IPS
> 
> NVIDIA Geforce (RTX) 2070 : nvidia-smi [8GB - cuda cores: 2304] Compute capability = 7.5 ((Turing))//see https://developer.nvidia.com/cuda-gpus AND https://en.wikipedia.org/wiki/CUDA sm_75
> 
> First error: terminate called after throwing an instance of 'ml::MLibException' what(): process:46: an illegal memory access was encountered:process if I start and start again: /media/hd2/Lab/SLAM/BundleFusion_Ubuntu_Pangolin/src/SiftGPU/SiftPyramid.cpp(387) : cudaSafeCall() Runtime API error 700: an illegal memory access was encountered.

My laptop's GPU is GTX1050, I have checked `CUDA_NVCC_FLAGS`  in CMakeLists.txt, but there still occurs an error:
```
terminate called after throwing an instance of 'ml::MLibException'
  what():  process:46: an illegal memory access was encountered:process
```
I think the problem is same with yours. Have you solved this?

busybeaver42, 2022-08-09:
No, is still not working. Perhaps my NVIDIA Geforce (RTX) 2070 : nvidia-smi [8GB - cuda cores: 2304] has to less memory or BundleFusion need to much. But I don't see a possibility to increase my memory. On the other hand I don't see a way to reduce the memory consumption from BundleFusion. So I can't do anything. I have reached a blind alley.

sysu17363087, 2022-12-27:
> No, is still not working. Perhaps my NVIDIA Geforce (RTX) 2070 : nvidia-smi [8GB - cuda cores: 2304] has to less memory or BundleFusion need to much. But I don't see a possibility to increase my [...]

hi,did you fix this problem at last? I met the same problem:(

LSK0821, 2023-12-20:
I met the same problem, I use an RTX 30 with CUDA 11./home/liusikang/Code/BundleFusion_Ubuntu_V0/FriedLiver/Source/SiftGPU/SiftPyramid.cpp(387) : cudaSafeCall() Runtime API error 715: an illegal instruction was encountered.
Killed

LSK0821, 2023-12-20:
I met the same problem, I use RTX 30 with CUDA11.
/home/liusikang/Code/BundleFusion_Ubuntu_V0/FriedLiver/Source/SiftGPU/SiftPyramid.cpp(387) : cudaSafeCall() Runtime API error 715: an illegal instruction was encountered. Killed