niessner/BundleFusion #55: Crushed after running FriedLiver.exe (offline SensorDataReader), changed s_maxNumImages in zParametersBundlingDefault.txt

AmandaHuang66, 2020-05-24:
Hi, I changed the s_maxNumImage to 400 (from 1200) as suggested in https://blog.csdn.net/ffcjjhv/article/details/89221992

However, cmd still crushes after running FriedLiver.exe; if changed to any other number smaller than 400 (I tried 200 & 300), an error message such as:
[image] 

Any idea regarding if I can change the frame size to make them compatible or if the crush is simply a GPU out-of-space problem?

Thanks!

mshmoon, 2021-01-21:
The is s_maxNumImage is image number in your dataset.

tan-tao11, 2022-03-22:
Hi, have you solved the problem yet? I meet the seam problem as you.