niessner/BundleFusion #57: Error compiling - Cannot open include file: 'boost/serialization/array_wrapper.hpp'

l4m1a4, 2020-06-06:
Dear everybody,

I came across BundleFusion while watching "Two minutes Paper" and think it's very interesting.
I want to test BundleFusion on the sample dataset only (I'm no artist nor developers, just a dummy).
I think I followed this guide https://bericht.neopostmodern.com/posts/artist-guide-to-bundlefusion (without the KINECT part because I don't have KINECT, just want to compile and run one example).
I loaded the project sucessfully into VS2013 (with CUDA 7.0 integration) without errors or warnings, but when I tried compiling to Release mode, I got the following errors:
I tried google but to no avail, this seems that only I encountered this error.
Please give me advice or suggestion how to proceed, thank you very much.
It says BundleFusion-master\external\mlib\include\core-base/common.h(61): fatal error C1083: Cannot open include file: 'boost/serialization/array_wrapper.hpp': file not found, and I couldn't find this file at that location either. Am I missing something?
Note: I just tried to run the code for test the example only. 
My laptop doesn't have NVIDIA card, but for compilation of code, NVIDIA said it doesn't matter.
https://stackoverflow.com/questions/20186848/can-i-compile-a-cuda-program-without-having-a-cuda-device

1>------ Build started: Project: FriedLiver, Configuration: Release x64 ------
1>  Compiling CUDA source file Source\CUDACache.cu...
1>  
1>  D:\Downloads\BundleFusion-master\FriedLiver>"C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v7.0\bin\nvcc.exe" -gencode=arch=compute_35,code=\"sm_35,compute_35\" --use-local-env --cl-version 2013 -ccbin "D:\VS2013\VC\bin\x86_amd64" -rdc=true -IInclude\cutil\inc -I"D:\Downloads\BundleFusion-master\FriedLiver" -ISource\DXUT\Optional -ISource\DXUT\Core -IInclude\Uplink -I"C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v7.0\include" -I"C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v7.0\include"     --keep-dir x64\Release -maxrregcount=62  --machine 64 --compile -cudart static    -use_fast_math -DWIN32 -DNDEBUG -D_WINDOWS -DD3DXFX_LARGEADDRESS_HANDLE -D_CRT_SECURE_NO_WARNINGS -DNOMINMAX -D_UNICODE -DUNICODE -Xcompiler "/EHsc /W3 /nologo /Ox /Zi  /MT  " -o x64\Release\CUDACache.cu.obj "D:\Downloads\BundleFusion-master\FriedLiver\Source\CUDACache.cu" 
1>C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v7.0\include\cuda_runtime.h : warning C4819: The file contains a character that cannot be represented in the current code page (932). Save the file in Unicode format to prevent data loss
1>c:\program files\nvidia gpu computing toolkit\cuda\v7.0\include\cuda_runtime_api.h(1970): warning C4819: The file contains a character that cannot be represented in the current code page (932). Save the file in Unicode format to prevent data loss
1>c:\program files\nvidia gpu computing toolkit\cuda\v7.0\include\math_functions.h : warning C4819: The file contains a character that cannot be represented in the current code page (932). Save the file in Unicode format to prevent data loss
1>c:\program files\nvidia gpu computing toolkit\cuda\v7.0\include\math_functions.h(877): warning C4819: The file contains a character that cannot be represented in the current code page (932). Save the file in Unicode format to prevent data loss
1>c:\program files\nvidia gpu computing toolkit\cuda\v7.0\include\math_functions.h(1850): warning C4819: The file contains a character that cannot be represented in the current code page (932). Save the file in Unicode format to prevent data loss
1>c:\program files\nvidia gpu computing toolkit\cuda\v7.0\include\math_functions.h(2728): warning C4819: The file contains a character that cannot be represented in the current code page (932). Save the file in Unicode format to prevent data loss
1>c:\program files\nvidia gpu computing toolkit\cuda\v7.0\include\math_functions.h(3651): warning C4819: The file contains a character that cannot be represented in the current code page (932). Save the file in Unicode format to prevent data loss
1>c:\program files\nvidia gpu computing toolkit\cuda\v7.0\include\math_functions.h(4595): warning C4819: The file contains a character that cannot be represented in the current code page (932). Save the file in Unicode format to prevent data loss
1>c:\program files\nvidia gpu computing toolkit\cuda\v7.0\include\math_functions.h(5521): warning C4819: The file contains a character that cannot be represented in the current code page (932). Save the file in Unicode format to prevent data loss
1>c:\program files\nvidia gpu computing toolkit\cuda\v7.0\include\math_functions.h(6421): warning C4819: The file contains a character that cannot be represented in the current code page (932). Save the file in Unicode format to prevent data loss
1>c:\program files\nvidia gpu computing toolkit\cuda\v7.0\include\math_functions.h(7314): warning C4819: The file contains a character that cannot be represented in the current code page (932). Save the file in Unicode format to prevent data loss
1>c:\program files\nvidia gpu computing toolkit\cuda\v7.0\include\math_functions.h(8227): warning C4819: The file contains a character that cannot be represented in the current code page (932). Save the file in Unicode format to prevent data loss
1>c:\program files\nvidia gpu computing toolkit\cuda\v7.0\include\device_functions.h : warning C4819: The file contains a character that cannot be represented in the current code page (932). Save the file in Unicode format to prevent data loss
1>c:\program files\nvidia gpu computing toolkit\cuda\v7.0\include\device_functions.h(775): warning C4819: The file contains a character that cannot be represented in the current code page (932). Save the file in Unicode format to prevent data loss
1>c:\program files\nvidia gpu computing toolkit\cuda\v7.0\include\device_functions.h(1632): warning C4819: The file contains a character that cannot be represented in the current code page (932). Save the file in Unicode format to prevent data loss
1>c:\program files\nvidia gpu computing toolkit\cuda\v7.0\include\device_double_functions.h : warning C4819: The file contains a character that cannot be represented in the current code page (932). Save the file in Unicode format to prevent data loss
1>D:\Downloads\BundleFusion-master\external\mlib\include\core-base/common.h(61): fatal error C1083: Cannot open include file: 'boost/serialization/array_wrapper.hpp': No such file or directory
1>  CUDACache.cu
1>C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V120\BuildCustomizations\CUDA 7.0.targets(593,9): error MSB3721: The command ""C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v7.0\bin\nvcc.exe" -gencode=arch=compute_35,code=\"sm_35,compute_35\" --use-local-env --cl-version 2013 -ccbin "D:\VS2013\VC\bin\x86_amd64" -rdc=true -IInclude\cutil\inc -I"D:\Downloads\BundleFusion-master\FriedLiver" -ISource\DXUT\Optional -ISource\DXUT\Core -IInclude\Uplink -I"C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v7.0\include" -I"C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v7.0\include"     --keep-dir x64\Release -maxrregcount=62  --machine 64 --compile -cudart static    -use_fast_math -DWIN32 -DNDEBUG -D_WINDOWS -DD3DXFX_LARGEADDRESS_HANDLE -D_CRT_SECURE_NO_WARNINGS -DNOMINMAX -D_UNICODE -DUNICODE -Xcompiler "/EHsc /W3 /nologo /Ox /Zi  /MT  " -o x64\Release\CUDACache.cu.obj "D:\Downloads\BundleFusion-master\FriedLiver\Source\CUDACache.cu"" exited with code 2.
========== Build: 0 succeeded, 1 failed, 0 up-to-date, 0 skipped ==========

l4m1a4, 2020-06-06:
I downloaded boost 1.66 library and extract to my folder.
I included (I think) it in the Additional Library Directories in VS 2013, but still the same error. What could have I missed?

[image]

l4m1a4, 2020-06-06:
I included the Boost library path into VC++ Directories > Include Directories
and C/C++ > Additional Include Directories
and the above error is gone, but another error appears!

1>D:\VS2013\VC\include\concrt.h(5764): error C2001: newline in constant
1>tmpxft_000021cc_00000000-2_CUDACache.cudafe1.stub.c(4): fatal error C1004: unexpected end-of-file found
Where am I missing?

[image]
[image]

l4m1a4, 2020-06-06:
From my understanding, this error seems to be related to the source code and I think I'm completely stuck now.
A thread mentioned a similar error but with another code.
[https://github.com/colmap/colmap/issues/29](https://github.com/colmap/colmap/issues/29)

l4m1a4, 2020-06-06:
Here is the full error log

1>------ Build started: Project: FriedLiver, Configuration: Release x64 ------
1>  Compiling CUDA source file Source\CUDACache.cu...
1>  
1>  D:\Downloads\BundleFusion-master\FriedLiver>"C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v7.0\bin\nvcc.exe" -gencode=arch=compute_35,code=\"sm_35,compute_35\" --use-local-env --cl-version 2013 -ccbin "D:\VS2013\VC\bin\x86_amd64" -rdc=true -IInclude\cutil\inc -I"D:\Downloads\BundleFusion-master\FriedLiver" -ID:\Downloads\boost\boost_1_66_0 -ISource\DXUT\Optional -ISource\DXUT\Core -IInclude\Uplink -I"C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v7.0\include" -I"C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v7.0\include"     --keep-dir x64\Release -maxrregcount=62  --machine 64 --compile -cudart static    -use_fast_math -DWIN32 -DNDEBUG -D_WINDOWS -DD3DXFX_LARGEADDRESS_HANDLE -D_CRT_SECURE_NO_WARNINGS -DNOMINMAX -D_UNICODE -DUNICODE -Xcompiler "/EHsc /W3 /nologo /Ox /Zi  /MT  " -o x64\Release\CUDACache.cu.obj "D:\Downloads\BundleFusion-master\FriedLiver\Source\CUDACache.cu" 
1>C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v7.0\include\cuda_runtime.h : warning C4819: The file contains a character that cannot be represented in the current code page (932). Save the file in Unicode format to prevent data loss
1>c:\program files\nvidia gpu computing toolkit\cuda\v7.0\include\cuda_runtime_api.h(1970): warning C4819: The file contains a character that cannot be represented in the current code page (932). Save the file in Unicode format to prevent data loss
1>c:\program files\nvidia gpu computing toolkit\cuda\v7.0\include\math_functions.h : warning C4819: The file contains a character that cannot be represented in the current code page (932). Save the file in Unicode format to prevent data loss
1>c:\program files\nvidia gpu computing toolkit\cuda\v7.0\include\math_functions.h(877): warning C4819: The file contains a character that cannot be represented in the current code page (932). Save the file in Unicode format to prevent data loss
1>c:\program files\nvidia gpu computing toolkit\cuda\v7.0\include\math_functions.h(1850): warning C4819: The file contains a character that cannot be represented in the current code page (932). Save the file in Unicode format to prevent data loss
1>c:\program files\nvidia gpu computing toolkit\cuda\v7.0\include\math_functions.h(2728): warning C4819: The file contains a character that cannot be represented in the current code page (932). Save the file in Unicode format to prevent data loss
1>c:\program files\nvidia gpu computing toolkit\cuda\v7.0\include\math_functions.h(3651): warning C4819: The file contains a character that cannot be represented in the current code page (932). Save the file in Unicode format to prevent data loss
1>c:\program files\nvidia gpu computing toolkit\cuda\v7.0\include\math_functions.h(4595): warning C4819: The file contains a character that cannot be represented in the current code page (932). Save the file in Unicode format to prevent data loss
1>c:\program files\nvidia gpu computing toolkit\cuda\v7.0\include\math_functions.h(5521): warning C4819: The file contains a character that cannot be represented in the current code page (932). Save the file in Unicode format to prevent data loss
1>c:\program files\nvidia gpu computing toolkit\cuda\v7.0\include\math_functions.h(6421): warning C4819: The file contains a character that cannot be represented in the current code page (932). Save the file in Unicode format to prevent data loss
1>c:\program files\nvidia gpu computing toolkit\cuda\v7.0\include\math_functions.h(7314): warning C4819: The file contains a character that cannot be represented in the current code page (932). Save the file in Unicode format to prevent data loss
1>c:\program files\nvidia gpu computing toolkit\cuda\v7.0\include\math_functions.h(8227): warning C4819: The file contains a character that cannot be represented in the current code page (932). Save the file in Unicode format to prevent data loss
1>c:\program files\nvidia gpu computing toolkit\cuda\v7.0\include\device_functions.h : warning C4819: The file contains a character that cannot be represented in the current code page (932). Save the file in Unicode format to prevent data loss
1>c:\program files\nvidia gpu computing toolkit\cuda\v7.0\include\device_functions.h(775): warning C4819: The file contains a character that cannot be represented in the current code page (932). Save the file in Unicode format to prevent data loss
1>c:\program files\nvidia gpu computing toolkit\cuda\v7.0\include\device_functions.h(1632): warning C4819: The file contains a character that cannot be represented in the current code page (932). Save the file in Unicode format to prevent data loss
1>c:\program files\nvidia gpu computing toolkit\cuda\v7.0\include\device_double_functions.h : warning C4819: The file contains a character that cannot be represented in the current code page (932). Save the file in Unicode format to prevent data loss
1>C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v7.0\include\cuda_runtime_api.h(1970): warning C4819: The file contains a character that cannot be represented in the current code page (932). Save the file in Unicode format to prevent data loss
1>C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v7.0\include\device_functions.h : warning C4819: The file contains a character that cannot be represented in the current code page (932). Save the file in Unicode format to preven [... cut ...] IA GPU Computing Toolkit\CUDA\v7.0\include"     --keep-dir x64\Release -maxrregcount=62  --machine 64 --compile -cudart static    -use_fast_math -DWIN32 -DNDEBUG -D_WINDOWS -DD3DXFX_LARGEADDRESS_HANDLE -D_CRT_SECURE_NO_WARNINGS -DNOMINMAX -D_UNICODE -DUNICODE -Xcompiler "/EHsc /W3 /nologo /Ox /Zi  /MT  " -o x64\Release\CUDACache.cu.obj "D:\Downloads\BundleFusion-master\FriedLiver\Source\CUDACache.cu"" exited with code 2.
========== Build: 0 succeeded, 1 failed, 0 up-to-date, 0 skipped ==========

l4m1a4, 2020-06-06:
FYI: 
I changed my Windows Language and locale and Language for non-Unicode Programs to ENglish like in this thread and not it's compiling. The above errors are gone. Waiting to see if there are new errors! Wish me luck.
https://stackoverflow.com/questions/10501634/warning-c4819-how-to-find-the-character-that-has-to-be-saved-in-unicode

l4m1a4, 2020-06-06:
OMG! 
========== Build: 1 succeeded, 0 failed, 0 up-to-date, 0 skipped ==========
I'll mark this thread solved and then try testing the EXE to see if it can run on my machine or not.
PHew, what a long day!

closed 2020-06-06 by l4m1a4

Byastal, 2020-10-09:
> Here is the full error log
>
> 1>------ Build started: Project: FriedLiver, Configuration: Release x64 ------
> 1> Compiling CUDA source file Source\CUDACache.cu...
> 1>
> 1> [...]

Excuse me, I am also very upset about this problem. Could you share your solution？