niessner/BundleFusion #59: ZLib_Ofcep.dll missing when running with debug mode

KevinLee752, 2020-06-28:
Hi, did anyone know about the error "Cannot find ZLib_Ofcep.dll" when running this code with debug mode? I got this error when running this code on my computer with win10 + VS 2013 + CUDA 10 + RTX 2070, and with debug mode (release mode works fine). I searched the “ZLib_Ofcep.dll” online but got nothing matched. Any advice would be  very appreciated.

Lazy-poe, 2020-09-14:
I meet the same problem.   Did you solve it？

KevinLee752, 2020-10-14:
@Lazy-poe No, not solved yet. I also met this problem when running Colmap. So sad....，Did you solve it?

zzdengke, 2022-02-07:
I met the same problem, how did you finally solve it?

E-Devil, 2023-06-21:
@KevinLee752 it's 3 years later, i also met this issue when debugging colmap; i solve this issue by copy the missing dll from release version