niessner/BundleFusion #63: ply blurry

flyisme, 2020-07-31:
win10(cuda10.2 )+GeForce GTX 1060+vs 2017+ intelD415

[image]

flyisme, 2020-07-31:
Is there a problem with the parameter configuration? How should I deal with this? Any suggestions? Thank you

tyronedong, 2020-08-01:
Glad to see you are running this on d415. I am using d435 but haven't run seccessfully. Could you please share your d415 driver code. It would be great if I could have you WeChat.

flyisme, 2020-08-03:
> Glad to see you are running this on d415. I am using d435 but haven't run seccessfully. Could you please share your d415 driver code. It would be great if I could have you WeChat.

Configure the 2.0sdk.

`

#include "stdafx.h"
#include "librealsense2/rs.hpp"
#include "RealSenseSensor2d0.h"

#ifdef REAL_SENSE

RealSenseSensor2d0::RealSenseSensor2d0()
{
	//d515
	/*unsigned int depthWidth = 1024; //! todo
	unsigned int depthHeight = 768;
	unsigned int colorWidth = 1920;
	unsigned int colorHeight = 1080;*/

	//d435i d415
	unsigned int depthWidth = 1280; //! todo
	unsigned int depthHeight = 720;
	unsigned int colorWidth = 1920;
	unsigned int colorHeight = 1080;

	
	

	RGBDSensor::init(depthWidth, depthHeight, colorWidth, colorHeight);

	//RGB
	cfg.enable_stream(RS2_STREAM_COLOR, colorWidth, colorHeight, RS2_FORMAT_RGB8, 30);
	//DEPTH
	cfg.enable_stream(RS2_STREAM_DEPTH, depthWidth, depthHeight, RS2_FORMAT_Z16, 30);
}

RealSenseSensor2d0::~RealSenseSensor2d0()
{

}

void RealSenseSensor2d0::createFirstConnected()
{
	// Start streaming with the default recommended configuration
	rs2::pipeline_profile pipe_profile =pipe.start(cfg);

	const int fisheye_sensor_idx = 1;

	rs2::stream_profile fisheye_stream = pipe_profile.get_stream(RS2_STREAM_DEPTH, 0);
	rs2_intrinsics intrinsics = fisheye_stream.as<rs2::video_stream_profile>().get_intrinsics();
	
	printf("depth:fx:%f,fy:%f\n", intrinsics.fx,intrinsics.fy);

	//Éî¶ÈÏà»úÄÚ²Î
	initializeDepthIntrinsics(intrinsics.fx, intrinsics.fy, getDepthWidth() / 2.f, getDepthHeight() / 2.f);

	fisheye_stream = pipe_profile.get_stream(RS2_STREAM_COLOR, 0);
	intrinsics = fisheye_stream.as<rs2::video_stream_profile>().get_intrinsics();

	//RGBÏà»úÄÚ²Î
	initializeColorIntrinsics(intrinsics.fx, intrinsics.fy, getColorWidth() / 2.f, getColorHeight() / 2.f);

	printf("color:fx:%f,fy:%f\n", intrinsics.fx, intrinsics.fy);

	//Ïà»úÍâ²Î
	initializeColorExtrinsics(mat4f::identity());
	initializeDepthExtrinsics(mat4f::identity());
}

bool RealSenseSensor2d0::processDepth()
{
	rs2::frameset data = pipe.wait_for_frames(); // Wait for next set of frames from the camera
	rs2::depth_frame depth = data.get_depth_frame();
	rs2::video_frame color = data.get_color_frame();            // Find the color data

	/*rs2::colorizer c;
	color = c.colorize(depth);*/

	rs2_error* e = 0;

	printf("\nframe num:%d,",data.get_frame_number());

	float* depth_f = getDepthFloat();
	//float* depth_f  =new float[getDepthWidth()*getDepthHeight()];

	//const uint16_t* depth_frame_data = (const uint16_t*)(rs2_get_frame_data(depth.get(), &e));

	//const uint16_t* depth_frame_data = (const uint16_t*)(depth.get_data());

	vec4uc* color_v = getColorRGBX();
	//vec4uc* color_v= new vec4uc[getColorWidth()*getColorHeight()];

	int n = getColorHeight();

	
	const uint8_t* rgb_frame_data = (const uint8_t*)(color.get_data());
	int stride = color.get_stride_in_bytes();

	for (int i = 0; i < (int)getColorWidth(); i++)
	{
		for (int j = 0; j < (int)getColorHeight(); j++) {
			
			//R
			int R= rgb_frame_data[j * stride + (i * 3)];
			//G
			int G=rgb_frame_data[j * stride + (i * 3) + 1];
			//B
			int B=rgb_frame_data[j * stride + (i * 3) + 2];
			color_v[i+j* getColorWidth()] = vec4uc(R,G, B, 255);
		}
	}

	printf("ÖÐÐÄµã¾àÀë£º%f\n", depth.get_distance(getDepthWidth()/2, getDepthHeight()/2));
	

	for (int x = 0; x < getDepthWidth(); x++)
	{
		for (int y = 0; y < getDepthHeight(); y++)
		{
			//const USHORT& d = *depth_frame_data++;

			depth_f[x + y * getDepthWidth()]= depth.get_distance(x,y);
			
			/*if (d == 0)
			{
				depth_f[x + y * getDepthWidth()] = 0.0f;
			}
			else
			{
				depth_f[x + y * getDepthHeight()] = (float)d;
			}*/
			
		}
	}

	
	

	// depth
	//memcpy(m_depthFloat[m_currentRingBufIdx], depth_f, sizeof(float)*getDepthWidth()*getDepthHeight());

	// color
	//memcpy(m_colorRGBX, color_v, sizeof(vec4uc)*getColorWidth()*getColorHeight());
	return true;
}

#endif
`

tyronedong, 2020-08-03:
> 
> 
> > Glad to see you are running this on d415. I am using d435 but haven't run seccessfully. Could you please share your d415 driver code. It would be great if I could have you WeChat.
> 
> Configure the 2.0sdk.
> 
> ```c++
> #include "stdafx.h"
> #include "librealsense2/rs.hpp"
> #include "RealSenseSensor2d0.h"
> 
> 
> 
> 
> #ifdef REAL_SENSE
> 
> 
> RealSenseSensor2d0::RealSenseSensor2d0()
> {
> 	//d515
> 	/*unsigned int depthWidth = 1024; //! todo
> 	unsigned int depthHeight = 768;
> 	unsigned int colorWidth = 1920;
> 	unsigned int colorHeight = 1080;*/
> 
> 	//d435i d415
> 	unsigned int depthWidth = 1280; //! todo
> 	unsigned int depthHeight = 720;
> 	unsigned int colorWidth = 1920;
> 	unsigned int colorHeight = 1080;
> 
> 
> 
> 	
> 	
> 
> 	RGBDSensor::init(depthWidth, depthHeight, colorWidth, colorHeight);
> 
> 
> 	//RGB
> 	cfg.enable_stream(RS2_STREAM_COLOR, colorWidth, colorHeight, RS2_FORMAT_RGB8, 30);
> 	//DEPTH
> 	cfg.enable_stream(RS2_STREAM_DEPTH, depthWidth, depthHeight, RS2_FORMAT_Z16, 30);
> }
> 
> RealSenseSensor2d0::~RealSenseSensor2d0()
> {
> 
> 
> }
> 
> void RealSenseSensor2d0::createFirstConnected()
> {
> 	// Start streaming with the default recommended configuration
> 	rs2::pipeline_profile pipe_profile =pipe.start(cfg);
> 
> 	const int fisheye_sensor_idx = 1;
> 
> 	rs2::stream_profile fisheye_stream = pipe_profile.get_stream(RS2_STREAM_DEPTH, 0);
> 	rs2_intrinsics intrinsics = fisheye_stream.as<rs2::video_stream_profile>().get_intrinsics();
> 	
> 	printf("depth:fx:%f,fy:%f\n", intrinsics.fx,intrinsics.fy);
> 
> 	//Éî¶ÈÏà»úÄÚ²Î
> 	initializeDepthIntrinsics(intrinsics.fx, intrinsics.fy, getDepthWidth() / 2.f, getDepthHeight() / 2.f);
> 
> 
> 	fisheye_stream = pipe_profile.get_stream(RS2_STREAM_COLOR, 0);
> 	intrinsics = fisheye_stream.as<rs2::video_stream_profile>().get_intrinsics();
> 
> 	//RGBÏà»úÄÚ²Î
> 	initializeColorIntrinsics(intrinsics.fx, intrinsics.fy, getColorWidth() / 2.f, getColorHeight() / 2.f);
> 
> 	printf("color:fx:%f,fy:%f\n", intrinsics.fx, intrinsics.fy);
> 
> 
> 	//Ïà»úÍâ²Î
> 	initializeColorExtrinsics(mat4f::identity());
> 	initializeDepthExtrinsics(mat4f::identity());
> }
> 
> bool RealSenseSensor2d0::processDepth()
> {
> 	rs2::frameset data = pipe.wait_for_frames(); // Wait for next set of frames from the camera
> 	rs2::depth_frame depth = data.get_depth_frame();
> 	rs2::video_frame color = data.get_color_frame();            // Find the color data
> 
> 	/*rs2::colorizer c;
> 	color = c.colorize(depth);*/
> 
> 	rs2_error* e = 0;
> 
> 
> 	printf("\nframe num:%d,",data.get_frame_number());
> 
> 	float* depth_f = getDepthFloat();
> 	//float* depth_f  =new float[getDepthWidth()*getDepthHeight()];
> 
> 	//const uint16_t* depth_frame_data = (const uint16_t*)(rs2_get_frame_data(depth.get(), &e));
> 
> 	//const uint16_t* depth_frame_data = (const uint16_t*)(depth.get_data());
> 
> 	vec4uc* color_v = getColorRGBX();
> 	//vec4uc* color_v= new vec4uc[getColorWidth()*getColorHeight()];
> 
> 	int n = getColorHeight();
> 
> 	
> 	const uint8_t* rgb_frame_data = (const uint8_t*)(color.get_data());
> 	int stride = color.get_stride_in_bytes();
> 
> 	for (int i = 0; i < (int)getColorWidth(); i++)
> 	{
> 		for (int j = 0; j < (int)getColorHeight(); j++) {
> 			
> 			//R
> 			int R= rgb_frame_data[j * stride + (i * 3)];
> 			//G
> 			int G=rgb_frame_data[j * stride + (i * 3) + 1];
> 			//B
> 			int B=rgb_frame_data[j * stride + (i * 3) + 2];
> 			color_v[i+j* getColorWidth()] = vec4uc(R,G, B, 255);
> 		}
> 	}
> 
> 	printf("ÖÐÐÄµã¾àÀë£º%f\n", depth.get_distance(getDepthWidth()/2, getDepthHeight()/2));
> 	
> 
> 	for (int x = 0; x < getDepthWidth(); x++)
> 	{
> 		for (int y = 0; y < getDepthHeight(); y++)
> 		{
> 			//const USHORT& d = *depth_frame_data++;
> 
> 			depth_f[x + y * getDepthWidth()]= depth.get_distance(x,y);
> 			
> 			/*if (d == 0)
> 			{
> 				depth_f[x + y * getDepthWidth()] = 0.0f;
> 			}
> 			else
> 			{
> 				depth_f[x + y * getDepthHeight()] = (float)d;
> 			}*/
> 			
> 		}
> 	}
> 
> 
> 
> 	
> 	
> 
> 	// depth
> 	//memcpy(m_depthFloat[m_currentRingBufIdx], depth_f, sizeof(float)*getDepthWidth()*getDepthHeight());
> 
> 	// color
> 	//memcpy(m_colorRGBX, color_v, sizeof(vec4uc)*getColorWidth()*getColorHeight());
> 	return true;
> }
> 
> #endif
> ```

Thanks for your sharing! Now I have a new problem. I compile BundleFusion with vs2013, but librealsense2 need vs2015 or higher. When I switch to vs2015, I have a compile problem same as `https://github.com/niessner/mLib/issues/11` . 
It is [image]
Have you met the same problem? If not, I will let you know when I successfully run with my D435 and see if ply blurry or not.

flyisme, 2020-08-04:
> Thanks for your sharing! Now I have a new problem. I compile BundleFusion with vs2013, but librealsense2 need vs2015 or higher. When I switch to vs2015, I have a compile problem same as [https://github.com/niessner/mLib/issues/11](url)
> It is [image]
> Have you met the same problem? If not, I will let you know when I successfully run with my D435 and see if ply blurry or not.

I did not encounter this problem, it may be that one of my dependent libraries is relatively old

flyisme, 2020-08-04:
other question 
[image]

`getHeapFreeCount()` 
It is always decreasing most of the time, and finally there is not enough memory，I just modified "SDF Voxel Size = 0.004f;"

chethanab16, 2020-09-18:
@flyisme @tyronedong ,**I am Trying  to compile Bundle fusion with win10(cuda10.2 +vs 2017, but I am Getting the following Error ,**
[image]

**I also Tried using the new mlib But still got the error as follows ,Could some one give me the suggestion how to fix this issue** **Thankyou**
[image]

lombardm, 2020-09-18:
> 
> 
> other question
> [image]
> 
> `getHeapFreeCount()`
> It is always decreasing most of the time, and finally there is not enough memory，I just modified "SDF Voxel Size = 0.004f;"

Hi @flyisme , did you manage to solve this? I am encoutering the same problem. Everything works fine with default settings but when I try do decrease the voxel size it crashes after two hundreds of frames. I also tried to modify s_hashNumBuckets in zPrametersDefault but without success!

chethanab16, 2020-09-24:
Hi Any one successfully compiled Bundle Fusion with VS 2017

BruceWANGDi, 2020-10-12:
@chethanab16  
#57 is helpful
The reason why the error described on your second photo occurred is that you lose the document 'array_wrapper.hpp', you should put this document in your library(the location is \mLibExternal\include\boost\serialization), you can find 'array_wrapper' by searching its name on google.

BruceWANGDi, 2020-10-12:
@flyisme 
Hi, have you figured out any solution to improve resolution? The .ply document is looked very blurry.

chethanab16, 2020-10-12:
@BruceWANGDI  thankyoufor providing me the link, yes as you mention
array_wrappe  is missing  and in issue #57  they specifide it for VS 2013
which works fine but when i try to add the array_wrapper.hpp to the
location  \mLibExternal\include\boost\serialization then again I will
receive the following error

Stoppable911222, 2020-12-09:
Align the depth and color could help imporve the resolution

H-tr, 2023-11-11:
> win10(cuda10.2 )+GeForce GTX 1060+vs 2017+ intelD415
> 
> [image]

Hi, have you figured out the reason? I'm facing the same problem using Realsense D435. I'm wondering if it is a problem with the setting or camera input quality is not good.