niessner/BundleFusion #70: vs2019 and cuda10.2 occurs C2280 errors in SolveTriangular.h 189

guyuezuntinggithub, 2021-06-09:
hi guys.
vs2019 and cuda10.2 occurs C2280 errors in SolveTriangular.h
C2280	“Eigen::Block<Eigen::Block<Eigen::Map<Derived,0,Eigen::Stride<0,0>>,-1,-1,false>,-1,-1,false> &Eigen::Block<Eigen::Block<Eigen::Map<Derived,0,Eigen::Stride<0,0>>,-1,-1,false>,-1,-1,false>::operator =(const Eigen::Block<Eigen::Block<Eigen::Map<Derived,0,Eigen::Stride<0,0>>,-1,-1,false>,-1,-1,false> &)”: “attempting to reference a deleted function”

Havehandssook, 2021-06-29:
Hello, I have the same problem. Have you solved it now?

yan-zeng-quan, 2021-08-01:
You need to find the Macros.h and replace 
"#if defined(_MSC_VER) && (!defined(__INTEL_COMPILER))" 
by 
"#if defined(_MSC_VER) && (_MSC_VER < 1900) && (!defined(__INTEL_COMPILER))"

wuqs-net, 2022-01-12:
Another way that seems to work is to download Eigen 3.2.10 and replace the mLibExternal/Eigen directory with the one in version 3.2.10.