niessner/BundleFusion #72: VS 2019 and CUDA 11.4

jianghd1996, 2021-12-03:
Except the same issue in https://github.com/niessner/BundleFusion/issues/70 (I could not find the corresponding code in solution now)

also 

C2280	“ml::MeshData<float> &ml::MeshData<float>::operator =(const ml::MeshData<float> &)”: 
FriedLiver C:\Users\alw\Desktop\冬奥\BundleFusion-master\external\mlib\include\core-mesh\meshData.cpp	575

wuqs-net, 2022-01-20:
Hello. Were you able to compile and run the project successfully? I was able to compile but then when I run the program with offline data it just got stuck.

crud89, 2022-03-30:
> Hello. Were you able to compile and run the project successfully? I was able to compile but then when I run the program with offline data it just got stuck. I am also using VS 2019 and CUDA 11.4.

Either update the `mLib` submodule from command line (it got fixed, but the submodule is still on an older revision):

```
cd external/mLib/
git pull origin master
```

Or manually edit the `meshData.h` file. For this look for the constructor definition at line *402* and add a compiler-generated override for the copy constructor and operator:

```cxx
MeshData(const MeshData&) = default;
void operator=(const MeshData&) = default;
```

zouhongwei, 2022-08-10:
> Hello. Were you able to compile and run the project successfully? I was able to compile but then when I run the program with offline data it just got stuck. I am also using VS 2019 and CUDA 11.4.

hello, have you fixed the problem?